oe-alliance/oe-alliance-core #89: Error when fetch from svn.code.sf.net

kingvuplus, 2015-07-17:
all file from http://sourceforge.net not work 

Fetcher failure: Fetch command failed with exit code 1, output:
svn: E000111: Unable to connect to a repository at URL 'svn://svn.code.sf.net/p/openpli/code/trunk/external/tuxcom'
svn: E000111: Can't connect to host 'svn.code.sf.net': Connection refused

atvcaptain, 2015-07-17:
yes must wait sf.net goes online now in recovery

kingvuplus, 2015-07-17:
ok thank you

opendroid-Team, 2015-07-19:
From what I understand is not a simple thing

cybertux, 2015-07-19:
Maybe one option could be setting up mirrors of those repositories and push them to github. Or use [PREMIRROR](http://patchwork.openembedded.org/patch/29799/)   variable.

graugans, 2015-07-19:
I've pushed some mirrors from repos I had lying arround on my disk:
 - https://bitbucket.org/pb-powerboard/gst-plugin-dvbmediasink.git
 - https://bitbucket.org/pb-powerboard/gstsubsink.git
 - https://bitbucket.org/pb-powerboard/hotplug-e2-helper.git
 - https://bitbucket.org/pb-powerboard/plugin-ambx.git
 - https://bitbucket.org/pb-powerboard/tuxtxt.git

Schimmelreiter, 2015-07-19:
It seems they have finally moved to Github:
https://github.com/OpenPLi

closed 2015-10-16 by koivo