oe-alliance/oe-alliance-core #99: wetek play image is not baking anymore

justason, 2015-12-26:
crashes on enigma2 recipe.

koivo, 2015-12-27:
Yes same here:
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for openhdf-base:
 * 	libungif4 (>= 4.1.4) *

atvcaptain, 2015-12-27:
hdf distro.conf is old

compare this

https://github.com/oe-alliance/oe-alliance-core/blob/3.2/meta-oe/conf/distro/openhdf.conf#L80

https://github.com/oe-alliance/oe-alliance-core/blob/3.2/meta-oe/conf/distro/openatv.conf#L82

justason, 2015-12-27:
Im backing OpenAtv.
crashed on enigna2.bb do_configure stage.

atvcaptain, 2015-12-27:
use wront buidl atv you must use this: https://github.com/wetek-enigma/build-enviroment

atvcaptain, 2015-12-27:
wetek not update in oe-a build

henrylicious, 2015-12-30:
wetek is updated now...

closed 2016-01-28 by madieatv