oe-alliance/oe-alliance-core #116: Error when build For DM800

kingvuplus, 2016-03-12:
i got this Error with Cairo When build for DM800 HD 

ERROR: Function failed: do_compile (log file is located at /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/temp/log.do_compile.21146)
ERROR: Logfile of failure stored in: /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/temp/log.do_compile.21146
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-32', 'mips-common', 'common-linux', 'common-glibc', 'mipsel-linux', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 2
| Makefile:1489: warning: overriding commands for target `recheck'
| Makefile:1127: warning: ignoring old commands for target `recheck'
| make  all-recursive
| make[1]: Entering directory `/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/build'
| Makefile:1489: warning: overriding commands for target `recheck'
| Makefile:1127: warning: ignoring old commands for target `recheck'
| Making all in src
| make[2]: Entering directory `/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/build/src'
| make  all-am
| make[3]: Entering directory `/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/build/src'
| ../mipsel-oe-linux-libtool  --tag=CC   --mode=compile mipsel-oe-linux-gcc  -mel -mabi=32 -msoft-float -march=mips32 --sysroot=/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800 -DHAVE_CONFIG_H -I. -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src -I..  -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/pixman-1    -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/freetype2 -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/freetype2  -D_REENTRANT -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/directfb -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/libpng15                      -Os -pipe -g -feliminate-unused-debug-types -ffat-lto-objects -c -o cairo-analysis-surface.lo /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairo-analysis-surface.c
| ../mipsel-oe-linux-libtool  --tag=CC   --mode=compile mipsel-oe-linux-gcc  -mel -mabi=32 -msoft-float -march=mips32 --sysroot=/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800 -DHAVE_CONFIG_H -I. -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src -I..  -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/pixman-1    -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/freetype2 -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/freetype2  -D_REENTRANT -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/directfb -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/libpng15                      -Os -pipe -g -feliminate-unused-debug-types -ffat-lto-objects -c -o cairo-arc.lo /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairo-arc.c
| mipsel-oe-linux-libtool: compile:  mipsel-oe-linux-gcc -mel -mabi=32 -msoft-float -march=mips32 --sysroot=/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800 -DHAVE_CONFIG_H -I. -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src -I.. -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/pixman-1 -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/freetype2 -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/freetype2 -D_REENTRANT -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/directfb -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/libpng15 -Os -pipe -g -feliminate-unused-debug-types -ffat-lto-objects -c /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairo-analysis-surface.c  -fPIC -DPIC -o .libs/cairo-analysis-surface.o
| mipsel-oe-linux-libtool: compile:  mipsel-oe-linux-gcc -mel -mabi=32 -msoft-float -march=mips32 --sysroot=/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800 -DHAVE_CONFIG_H -I. -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src -I.. -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/pixman-1 -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/freetype2 -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/freetype2 -D_REENTRANT -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/directfb -I/home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/sysroots/dm800/usr/include/libpng15 -Os -pipe -g -feliminate-unused-debug-types -ffat-lto-objects -c /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairo-arc.c  -fPIC -DPIC -o .libs/cairo-arc.o
| In file included from /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairo-mutex-type-private.h:45:0,
|                  from /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairo-scaled-font-private.h:45,
|                  from /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairoint.h:422,
|                  from /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairo-analysis-surface.c:37:
| /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairo-mutex-impl-private.h:262:3: error: #error "XXX: No mutex implementation found.  Cairo will not work with multiple threads.  Define CAIRO_NO_MUTEX to 1 to acknowledge and accept this limitation and compile cairo without thread-safety support."
|  # error "XXX: No mutex implementation found.  Cairo will not work with multiple threads.  Define CAIRO_NO_MUTEX to 1 to acknowledge and accept this limitation and compile cairo without thread-safety support."
|    ^
| In file included from /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairo-mutex-type-private.h:45:0,
|                  from /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/cairo-1.14.2/src/cairo-scaled-font-private.h:45,
[... cut ...]
| WARNING: /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/temp/run.do_compile.21146:1 exit 1 from
|   exit 1
| ERROR: Function failed: do_compile (log file is located at /home/sodo/Bureau/fed/build-enviroment/builds/openatv/release/dm800/tmp/work/mips32el-nf-oe-linux/cairo/1.14.2-r0/temp/log.do_compile.21146)
ERROR: Task 9093 (/home/sodo/Bureau/fed/build-enviroment/openembedded-core/meta/recipes-graphics/cairo/cairo_1.14.2.bb, do_compile) failed with exit code '1'

Any One can Fix That Thank you

atvcaptain, 2016-03-12:
for dm800 you must use oe-a 2.3 build all other not support the old kernel

kingvuplus, 2016-03-12:
ok thank you

closed 2016-03-14 by arn354