oe-alliance/oe-alliance-core #128: IPTV does not work on OE Alliance images

Neustradamus, 2016-08-28:
Hi all,

IPTV does not work on OpenATV, OpenHDF, opendroid, OpenESI.

It works on dreambox-image, Merlin3, OoZooN, NewNigma2.

Can you look the bug?

Thanks in advance.

Regards,

Neustradamus

atvcaptain, 2016-08-29:
we use every time IPTV with ATV no Problem 

more infos , Box, Image Version, IPTV URL , what not working and and and

Neustradamus, 2016-08-29:
OK:
- http://feed.merlin3.info/feeds/images/Merlin-3_OE-2.0-dm7020hdV2-20151025.nfi
- http://dreamboxupdate.com/opendreambox/2.0.0/images/dm7020hd/dreambox-image-dm7020hdv2-20160701.nfi
- http://oozoon-download.de/opendreambox/images/dm7020hdv2/OoZooN-Image-dm7020hdv2-20160117.nfi
- http://feed.newnigma2.to/daily/images/archive/dm7020hdv2/newnigma2-ipk-daily-dm7020hdv2-25_08_2016.nfi

NOK:
- http://images.mynonpublic.com/openatv/5.3/dm7020hdv2/openatv-5.3-dm7020hdv2-20160828.zip
- http://v55.hdfreaks.cc/dm7020hd/openhdf-5.5-dm7020hdv2-96-20160824.nfi
- http://images.opendroid.org/5.5/Dreambox/dm7020hdv2/opendroid-5.5-dm7020hdv2-20160817.nfi
- http://www.openesi.eu/images/images/Dreambox/dm7020hdv2/openesi-3.0-dm7020hdv2-20160629.nfi.zip

It's Free multiposte from France.
On OE-Alliance: No signal / on the firewall: no flow.

atvcaptain, 2016-08-29:
need urls from the streams to test it without more info we cant help, and you use plugin , build in or that for playing

Neustradamus, 2016-08-29:
But it is not free, only for Free customers with TV option.

atvcaptain, 2016-08-29:
so i cant help , no idea was wrong sorry

Neustradamus, 2016-08-29:
More informations on https://lafibre.info/free-tutoriels/multiposte-de-free-les-url-chaines-par-chaines/.

The OE must to request in RTSP (554/TCP) and the server sends in UDP and the OE sends in UDP too.

http://www.informit.com/articles/article.aspx?p=169578&seqNum=3

closed 2019-01-09 by arn354