oe-alliance/oe-alliance-core #131: Error Build In Compile FFmepg

kingvuplus, 2016-10-03:
| libavcodec/vc1_block.o: file not recognized: File truncated
| collect2: error: ld returned 1 exit status
| make: *** [libavcodec/libavcodec.so.57] Error 1
| WARNING: /home/sodovps/Bureau/oe/build-enviroment/builds/openeight/release/vusolo2/tmp/work/mips32el-oe-linux/ffmpeg/3.0.2-r4/temp/run.do_compile.3243:1 exit 1 from 'exit 1'
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (log file is located at /home/sodovps/Bureau/oe/build-enviroment/builds/openeight/release/vusolo2/tmp/work/mips32el-oe-linux/ffmpeg/3.0.2-r4/temp/log.do_compile.3243)
ERROR: Task 3068 (/home/sodovps/Bureau/oe/build-enviroment/meta-oe-alliance/meta-oe/recipes-multimedia/ffmpeg/ffmpeg_3.0.2.bb, do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2396 tasks of which 2393 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

arn354, 2016-10-03:
No issue here.
Is that a clean build from zero? Or a repeated build? Any local changes applied?
Have you tried a local PR bump of ffmpeg recipe?

kingvuplus, 2016-10-04:
fixed thank you

closed 2016-10-04 by arn354