oe-alliance/oe-alliance-core #142: {Request} OpenVPN Update

MMD70, 2016-11-24:
Hi,
Will OpenVPN become a update with the upcoming release of OpenVPN 2.4 in December?
Current version is quiet old.
Version 2.4 will bring AES-GCM which will substantially lower CPU cycles and off course bug fixes.
Then maybe wait until the audit:
https://ostif.org/ostif-is-beginning-a-fundraiser-for-openvpn-lets-get-it-audited/

MMD70, 2016-12-28:
Hi again,

Yesterday OpenVPN version 2.4 was released.
Will we see an update?

Thanks.

arn354, 2016-12-28:
if it is added to openembedded - yes.
if not - if someone provides a updated bitbake recipe for it to add it to oe-a - yes.

Schimmelreiter, 2016-12-28:
I'm working on adapting OpenVPN's easy-rsa to be less interactive ... preferably even non-interactive.
After I'm done doing that, I intend to make updates to OpenWebif, OpenVPN, OpenSSL and openvpn-easy-rsa at once.

There is no need to rush btw:
If new ciphers and stuff are supported, it's likely to break with certain clients, so give the clients time to upgrade as well.
Not all clients come directly from OpenVPN, e.g. SecurePoint SSL VPN, Tunnelblick, ...

MMD70, 2016-12-28:
@Schimmelreiter Will write again, great to see some is concerned with security, thanks :)
Personally I do not use easy-rsa but use XCA to generate certificates "crafted" for OpenVPN.
>>"If new ciphers and stuff are supported............"
If the server side is 2.4 and configured the right way that will not be a problem, see --ncp-ciphers in manual 2.4
For the client side is no problem at all.
https://community.openvpn.net/openvpn/wiki/Openvpn24ManPage

@arn354 Tomorrow, possibly tonight, I will test that ipk.
Thanks.

Schimmelreiter, 2016-12-28:
Actually even StrongSwan is on my To-Do list.
My main problem is the Python plugin part to configure the stuff.

I have absolutely no problem writing the bash scripts (or turning them into Python) to semi-automatically gather system info:
```
root@quadbox /etc/openvpn/easy-rsa # ./test
DYN.1   = mynet.dyn
DYN.2   = quadbox.dyn
DYN.3   = quadbox
DYN.4   = quadbox.box
DYN.5   = quadbox.local
IP.1    = 192.168.75.17
IP.2    = fdc1:affe:c0c0:0:4670:bff:fe12:3456
IP.3    = 2001:db8::4670:bff:fe12:3456
IP.4    = fe80::4670:bff:fe12:3456
```

This script gathers the hostnames and IPs of the box (I want the server certificate to work for OpenWebif too and thus I need to put all hostnames inside the cert under which the user might access OWIF), but I need user interaction, preferably inside the E2 GUI and not shell.

The user shall be able to add
- one or multiple "network DynDNS hosts" (DynDNS for IPv4 usually resolves to the router's IPv4, it's the same for all devices accessible through port forwarding/NAT) and/or 
- "per device DynDNS" (IPv6 DynDNS hosts resolve to the device's unique IPv6 address and thus are individual for each device, the same applies if you are in the lucky situation to have a public IPv4 subnet and can assign public IPv4 addresses to each/some of your devices).

I also want the user to be able to change Country, State and City for the corresponding Cert fields and username(s) for the client certs.

MMD70, 2016-12-28:
Looks like there`s more to do,
```
Installing openvpn (2.4.0-r0) on root.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for openvpn:
 *      libc6 (>= 2.24) *       libcrypto1.0.0 (>= 1.0.2j) *    libpam (>= 1.3.0) *     libssl1.0.0 (>= 1.0.2j) *
 * opkg_install: Cannot install package openvpn.
```
Unfortunately I don`t know what to do but I read in the "control" file:
`Depends: libc6 (>= 2.24), libcrypto1.0.0 (>= 1.0.2j), liblzo2-2 (>= 2.09), libpam (>= 1.3.0), libssl1.0.0 (>= 1.0.2j)` 
And in the bb file:
`DEPENDS = "lzo openssl libpam"`
Maybe adjust accordingly?

arn354, 2016-12-28:
I build in branch 4.0, so required packages should be there. Maybe not on your feed.

MMD70, 2016-12-28:
On feed there is opkg list | grep ****:
libc6 - 2.23-r0
libcrypto1.0.0 - 1.0.2g-r0.6
liblzo2-2 - 2.09-r0
libpam - 1.2.1-r5
libssl1.0.0 - 1.0.2g-r0.6
I`m "kind of naked" here :)
Makes me think OpenATV not have correct versions on feed?

arn354, 2016-12-29:
as said  - i have build here on branch 4.0 which will be openatv 6.
so it is normal that branch 4.0 has newer versions of libraries then branch 3.4 of openatv 5.3. Simply wait ;)

MMD70, 2016-12-29:
Ah yes, i see.

Will simply wait...:)

commit 7949a55b (2017-01-17): [openvpn] Bump to 2.4.0
Closes #142
closed 2017-01-17 by Schimmelreiter