oe-alliance/oe-alliance-core #149: Error building dvbsnoop (checksum failure)

ivars777, 2016-12-14:
Trying to build openatv using build-enviroment (4.0)
Constantly produce error while building dvbsnoop (checksum failure). Tried to change checksum in bb file, then apears unpacking error.

```
WARNING: dvbsnoop-1.4.50-r2 do_fetch: Renaming /build/build-enviroment/sources/dvbsnoop-1.4.50.tar.gz to /build/build-enviroment/sources/dvbsnoop-1.4.50.tar.gz_bad-checksum_bdee38f9f9843577ed1123ce433cf4cb
WARNING: dvbsnoop-1.4.50-r2 do_fetch: Checksum failure encountered with download of http://sourceforge.net/projects/dvbsnoop/files/latest/download/dvbsnoop-1.4.50.tar.gz - will attempt other sources if available
ERROR: dvbsnoop-1.4.50-r2 do_fetch: Fetcher failure for URL: 'http://sourceforge.net/projects/dvbsnoop/files/latest/download/dvbsnoop-1.4.50.tar.gz'. Checksum mismatch!
File: '/build/build-enviroment/sources/dvbsnoop-1.4.50.tar.gz' has md5 checksum bdee38f9f9843577ed1123ce433cf4cb when 68a5618c95b4372eea9ac5ec5005f299 was expected
File: '/build/build-enviroment/sources/dvbsnoop-1.4.50.tar.gz' has sha256 checksum 4e5fb42dc00b06e3bd90d442f73dc23aa14df4fe11e3bac16b88f9185d8c9b51 when 7658498b26a5d2a0242e81f0cfafa0e43a2bec56f8674e7ac197dfc310866ec6 was expected
If this change is expected (e.g. you have upgraded to a new version without updating the checksums) then you can use these lines within the recipe:
SRC_URI[md5sum] = "bdee38f9f9843577ed1123ce433cf4cb"
SRC_URI[sha256sum] = "4e5fb42dc00b06e3bd90d442f73dc23aa14df4fe11e3bac16b88f9185d8c9b51"
Otherwise you should retry the download and/or check with upstream to determine if the file has become corrupted or otherwise unexpectedly modified.

ERROR: dvbsnoop-1.4.50-r2 do_fetch: Fetcher failure for URL: 'http://sourceforge.net/projects/dvbsnoop/files/latest/download/dvbsnoop-1.4.50.tar.gz'. Unable to fetch URL from any source.
ERROR: dvbsnoop-1.4.50-r2 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /build/build-enviroment/builds/openatv/release/dm800sev2/tmp/work/mips32el-oe-linux/dvbsnoop/1.4.50-r2/temp/log.do_fetch.28768
ERROR: Task (/build/build-enviroment/meta-oe-alliance/meta-oe/recipes-multimedia/dvbsnoop/dvbsnoop.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 9197 tasks of which 19 didn't need to be rerun and 1 failed

```

arn354, 2016-12-14:
a change is in 3.4 - but not fully working for all boxes.
Simply wait :)

arn354, 2016-12-14:
should be ok

closed 2016-12-14 by arn354