oe-alliance/oe-alliance-core #162: Delay start OpenVPN

MMD70, 2017-01-25:
Hello,

The added autostart
https://github.com/oe-alliance/oe-alliance-core/blob/3.4/meta-oe/recipes-connectivity/openvpn/openvpn_2.4.0.bbappend
leads to failing connection attempt for OpenVPN because the system time is not yet synced.
This was also the case before the added autostart.
In the past and now I solve(d) that by adding a delay of 5 seconds before OpenVPN starts.

Maybe add this delay out of the box?
Thanks.

Schimmelreiter, 2017-02-01:
Ceterum censeo SysVinit esse delendam.

MMD70, 2017-02-01:
Putting the Latin together:
"For the rest, it is my opinion that SysVinit has to be destroyed"
sytemd then I guess :)

Schimmelreiter, 2017-02-02:
"Ceterum censeo Carthaginem esse delendam" or "Ceterum autem censeo Carthaginem esse delendam" (English: "Furthermore, (moreover) I consider that Carthage must be destroyed") ... according to several ancient sources the Roman statesman Cato the Elder frequently if not invariably ended his speeches in the Senate with a variant of this expression even when his speech had been totally unrelated to Roman foreign policy towards Carthage ...

Yes, SysVinit ... especially our highly limited variant of it ... must go.

You might have noticed that some of the startup scripts have additional comments at the very beginning:
```
#!/bin/sh
### BEGIN INIT INFO
# Provides:          ws4d
# Required-Start:    $network $remote_fs $syslog
# Required-Stop:     
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: WS-Discovery service
# Description:       Enables Windows machines to locate SMB shares on this machine
### END INIT INFO
```
These are not just comments, but they contained some info used by full versions of SysVinit to determine **when** to start this service.

For example:
As OpenVPN requires a valid time, you would add "$time" to the "Required-Start:" section to let SysVinit know about this requirement.

ntp on the other hand would have $time in provides ..

update-rc.d would then automagically add the OpenVPN start **after** ntp.

**Our** crippled SysVinit/update-rc.d however adds any service start at 20 in runlevels 2, 3, 4, 5 and stop at 20 in runlevels 0, 1 and 6, no matter what "Required-Start", "Required-Stop", "Provides", "Default-Start" and "Default-Stop" say.

So even a real SysVinit would help already, but the effort would probably be the same as switching to systemd.

For the moment, you can try this:
```
update-rc.d -f openvpn remove
update-rc.d -f openvpn start 98 2 3 4 5 . stop 10 0 1 6 .
```
That will let OpenVPN start much later and stop earlier.

If that helps, I'll check it in that way. I can not reproduce your problem and thus I can not test this cheat, because my boxes keep the time across reboots.

closed 2019-01-09 by arn354