oe-alliance/oe-alliance-core #163: USB tuner problem - ET10K, kernel upgraded to 4.8.3

ccs-ccs, 2017-01-30:
SUNDTEK USB TUNER CABLE TERRESTRIAL HD DVB-C/T/T2 works fine on OpenViX 4.2.026, but updating to 4.2.027  (which includes a kernel change to 4.8.3) causes the driver to fail to install and the box hangs.......
```
root@et10000:~#dmesg
[    1.255000] ehci-brcm ehci-brcm.3: Broadcom STB EHCI
[    1.255000] ehci-brcm ehci-brcm.3: new USB bus registered, assigned bus number 4
[    1.268000] ehci-brcm ehci-brcm.3: irq 72, io mem 0x10490500
[    1.275000] ehci-brcm ehci-brcm.3: USB 0.0 started, EHCI 1.00
[    1.283000] hub 4-0:1.0: USB hub found
[    1.283000] scsi 0:0:0:0: Direct-Access     ATA      HGST HTS541010A9 A560 PQ: 0 ANSI: 5
[    1.295000] hub 4-0:1.0: 1 port detected
[    1.300000] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    1.306000] ohci-brcm ohci-brcm.0: Broadcom STB OHCI
[    1.311000] ohci-brcm ohci-brcm.0: new USB bus registered, assigned bus number 5
[    1.319000] ohci-brcm ohci-brcm.0: irq 68, io mem 0x10480400
[    1.336000] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    1.342000] sd 0:0:0:0: [sda] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[    1.350000] sd 0:0:0:0: [sda] 4096-byte physical blocks
[    1.355000] sd 0:0:0:0: [sda] Write Protect is off
[    1.360000] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    1.360000] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    1.390000] hub 5-0:1.0: USB hub found
[    1.394000] hub 5-0:1.0: 1 port detected
[    1.398000] ohci-brcm ohci-brcm.1: Broadcom STB OHCI
[    1.403000] ohci-brcm ohci-brcm.1: new USB bus registered, assigned bus number 6
[    1.411000] ohci-brcm ohci-brcm.1: irq 69, io mem 0x10480600
[    1.483000] hub 6-0:1.0: USB hub found
[    1.487000] hub 6-0:1.0: 1 port detected
[    1.491000] ohci-brcm ohci-brcm.2: Broadcom STB OHCI
[    1.496000] ohci-brcm ohci-brcm.2: new USB bus registered, assigned bus number 7
[    1.504000] ohci-brcm ohci-brcm.2: irq 73, io mem 0x10490400
[    1.520000] usb 1-1: new high-speed USB device number 2 using ehci-brcm
[    1.576000] hub 7-0:1.0: USB hub found
[    1.580000] usb 3-1: new high-speed USB device number 2 using ehci-brcm
[    1.587000] hub 7-0:1.0: 1 port detected
[    1.591000] ohci-brcm ohci-brcm.3: Broadcom STB OHCI
[    1.596000] ohci-brcm ohci-brcm.3: new USB bus registered, assigned bus number 8
[    1.604000] ohci-brcm ohci-brcm.3: irq 74, io mem 0x10490600
[    1.661000] ata2: SATA link down (SStatus 0 SControl 300)
[    1.675000] hub 8-0:1.0: USB hub found
[    1.679000] hub 8-0:1.0: 1 port detected
[    1.684000] usbcore: registered new interface driver cdc_wdm
[    1.684000]  sda: sda1
[    1.684000] sd 0:0:0:0: [sda] Attached SCSI disk
[    1.697000] usbcore: registered new interface driver usb-storage
[    1.703000] mousedev: PS/2 mouse device common for all mice
[    1.709000] i2c /dev entries driver
[    1.712000] usbcore: registered new interface driver btusb
[    1.718000] sdhci: Secure Digital Host Controller Interface driver
[    1.724000] sdhci: Copyright(c) Pierre Ossman
[    1.729000] sdhci-pltfm: SDHCI platform and OF driver helper
[    1.734000] sdhci-brcmstb sdhci-brcmstb.0: Disabled by bootloader
[    1.740000] sdhci-brcmstb sdhci-brcmstb.1: Disabled by bootloader
[    1.747000] hidraw: raw HID events driver (C) Jiri Kosina
[    1.752000] usbcore: registered new interface driver usbhid
[    1.758000] usbhid: USB HID core driver
[    1.762000] NET: Registered protocol family 17
[    1.767000] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[    1.773000] Bluetooth: HIDP socket layer initialized
[    1.778000] Key type dns_resolver registered
[    1.779000] hub 3-1:1.0: USB hub found
[    1.779000] hub 3-1:1.0: 4 ports detected
[    1.790000] PM: CP0 COUNT/COMPARE frequency does not depend on divisor
[    1.801000] EBI CS1: setting up NAND flash (primary)
[    1.806000] nand: device found, Manufacturer ID: 0xec, Chip ID: 0xd3
[    1.813000] nand: Samsung NAND 1GiB 3,3V 8-bit
[    1.817000] nand: 1024 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[    1.825000] brcmnand brcmnand.0: 1024MiB total, 128KiB blocks, 2KiB pages, 16B OOB, 8-bit, BCH-16 (1KiB sector)
[    1.835000] Bad block table found at page 524224, version 0x01
[    1.841000] Bad block table found at page 524160, version 0x01
[    1.848000] Creating 3 MTD partitions on "brcmnand.0":
[... 107 lines cut ...]
[  253.586000] PrId  : 00025a11 (Broadcom BMIPS5000)
[  253.591000] Modules linked in: ipv6 rtl8192cu rtl_usb rtl8192c_common rtlwifi mac80211 xfs libcrc32c dvb(OF) modloader2(O) modloader(O) tpm(O)
[  253.604000] Process mediasrv (pid: 1070, threadinfo=cef20000, task=ced5a080, tls=74470930)
[  253.612000] Stack : cf64e410 cf801f68 cef21d30 cfcf0280 e11f9b80 e11f9b80 00000037 80938530
[  253.612000]    cf63a100 80940000 cf0781a0 80432690 cef21e38 00000000 00000001 cef21e38
[  253.612000]    cf63a100 807d0390 809c0000 cfcf0300 cf0781a0 cf63a100 00000000 cf944cc0
[  253.612000]    00000000 80137d14 cef21e38 cf94cbb0 cf63a100 8015955c 00000037 00000002
[  253.612000]    cf63a108 cf63a100 cf0781a0 00000000 cf63a108 80137c48 00000006 8012f34c
[  253.612000]    ...
[  253.649000] Call Trace:
[  253.652000] [<e0ae5108>] __67a1d0+0x150/0x1d0 [dvb]
[  253.657000] [<80432690>] misc_open+0x168/0x1e0
[  253.662000] [<80137d14>] chrdev_open+0xcc/0x1dc
[  253.666000] [<8012f34c>] do_dentry_open.isra.2+0x26c/0x468
[  253.672000] [<80142e9c>] path_openat+0x180/0x1134
[  253.676000] [<80145074>] do_filp_open+0xc8/0x120
[  253.681000] [<80130c08>] do_sys_open+0x1a0/0x280
[  253.686000] [<80015dd4>] syscall_common+0x34/0x58
[  253.690000]
[  253.692000]
[  253.692000] Code: 26140610  8ea20000  02802021 <8c450008> 0240f809  2406007c  3c02e0ae  2442468c  ae020634
[  253.702000] ---[ end trace 1966ad3d687ae7a8 ]---
root@et10000:~#

```

ccs-ccs, 2017-03-16:
ViX 5.0.003 has now been released and the problem still hasn't been sorted.

I've currently got £100 worth of tuners which don't work - the SundTek usb T2 tuner and a hybrid DVB-C/T2 tuner.
My ET10K has 4xDVB-C/T2 tuners. They all work independently, but 4 do not work at the same time.
All 4 tuners have individual feeds (no loop thru's) - if I remove the feed to tuner D, tuner C stops working!!!

Latest dmesg...

```
root@et10000:~# dmesg
[    1.022000] bcmgenet: configuring instance #0 for internal PHY
[    1.050000] bcmgenet bcmgenet.0 eth0: link down
[    1.054000] __clk_disable: enet [0]
[    1.054000] bcm40nm_pm_genet_disable 00
[    1.054000] __clk_disable: network [0]
[    1.054000] bcm7425_pm_network_disable 00
[    1.054000] __clk_enable: network [1]
[    1.054000] bcm7425_pm_network_enable 00
[    1.054000] __clk_enable: moca_genet [1]
[    1.054000] bcm40nm_pm_genet1_enable 00
[    1.054000] __clk_enable: moca [1]
[    1.054000] bcm40nm_pm_moca_enable 00
[    1.067000] Found PHY at Address 31
[    1.070000] bcmgenet: configuring instance #1 for external RGMII
[    1.077000] __clk_disable: moca [0]
[    1.077000] bcm40nm_pm_moca_disable 00
[    1.077000] __clk_disable: moca_genet [0]
[    1.077000] bcm40nm_pm_genet1_disable 00
[    1.077000] __clk_disable: network [0]
[    1.077000] bcm7425_pm_network_disable 00
[    1.077000] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    1.083000] __clk_enable: usb [1]
[    1.083000] bcm40nm_pm_usb_enable 00
[    1.083000] ehci-brcm ehci-brcm.0: Broadcom STB EHCI
[    1.088000] ehci-brcm ehci-brcm.0: new USB bus registered, assigned bus number 1
[    1.106000] ehci-brcm ehci-brcm.0: irq 66, io mem 0x10480300
[    1.119000] ehci-brcm ehci-brcm.0: USB 0.0 started, EHCI 1.00
[    1.125000] hub 1-0:1.0: USB hub found
[    1.129000] hub 1-0:1.0: 1 port detected
[    1.133000] ehci-brcm ehci-brcm.1: Broadcom STB EHCI
[    1.138000] ehci-brcm ehci-brcm.1: new USB bus registered, assigned bus number 2
[    1.156000] ehci-brcm ehci-brcm.1: irq 67, io mem 0x10480500
[    1.168000] ehci-brcm ehci-brcm.1: USB 0.0 started, EHCI 1.00
[    1.174000] hub 2-0:1.0: USB hub found
[    1.178000] hub 2-0:1.0: 1 port detected
[    1.182000] ehci-brcm ehci-brcm.2: Broadcom STB EHCI
[    1.187000] ehci-brcm ehci-brcm.2: new USB bus registered, assigned bus number 3
[    1.205000] ehci-brcm ehci-brcm.2: irq 71, io mem 0x10490300
[    1.205000] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    1.217000] ata1.00: ATA-8: HGST HTS541010A9E680, JA0OA560, max UDMA/133
[    1.217000] ata1.00: 1953525168 sectors, multi 16: LBA48 NCQ (depth 31/32)
[    1.230000] ata1.00: configured for UDMA/133
[    1.235000] scsi 0:0:0:0: Direct-Access     ATA      HGST HTS541010A9 A560 PQ: 0 ANSI: 5
[    1.237000] ehci-brcm ehci-brcm.2: USB 0.0 started, EHCI 1.00
[    1.249000] hub 3-0:1.0: USB hub found
[    1.253000] hub 3-0:1.0: 1 port detected
[    1.257000] ehci-brcm ehci-brcm.3: Broadcom STB EHCI
[    1.262000] ehci-brcm ehci-brcm.3: new USB bus registered, assigned bus number 4
[    1.280000] ehci-brcm ehci-brcm.3: irq 72, io mem 0x10490500
[    1.290000] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    1.293000] ehci-brcm ehci-brcm.3: USB 0.0 started, EHCI 1.00
[    1.302000] sd 0:0:0:0: [sda] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[    1.302000] hub 4-0:1.0: USB hub found
[    1.302000] hub 4-0:1.0: 1 port detected
[    1.302000] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    1.302000] ohci-brcm ohci-brcm.0: Broadcom STB OHCI
[    1.302000] ohci-brcm ohci-brcm.0: new USB bus registered, assigned bus number 5
[    1.302000] ohci-brcm ohci-brcm.0: irq 68, io mem 0x10480400
[    1.341000] sd 0:0:0:0: [sda] 4096-byte physical blocks
[    1.347000] sd 0:0:0:0: [sda] Write Protect is off
[... 130 lines cut ...]
[  538.492000] BadVA : 00000009
[  538.495000] PrId  : 00025a11 (Broadcom BMIPS5000)
[  538.500000] Modules linked in: ipv6 rtl8192cu rtl_usb rtl8192c_common rtlwifi mac80211 xfs libcrc32c dvb(OF) modloader2(O) modloader(O) tpm(O)
[  538.513000] Process mediasrv (pid: 1072, threadinfo=ce086000, task=ce040100, tls=73f70930)
[  538.521000] Stack : ce087d3c cfc9e610 ce087d38 cfcf4c80 e11f9b80 e11f9b80 00000037 8090c530
[  538.521000]    ce068900 80910000 cf6e4e90 8042e3d0 ce087e38 00000000 00000001 0000002f
[  538.521000]    ce068900 00000000 807ab000 cfcf4d00 cf6e4e90 ce068900 00000000 00000000
[  538.521000]    ce087e38 801363ac 00000006 80157b50 ce087e38 00000002 00000037 8013c230
[  538.521000]    ce068900 cf6e4e90 ce068900 ce068908 801362e4 8012dab8 808fa7b0 cf6e4e90
[  538.521000]    ...
[  538.557000] Call Trace:
[  538.560000] [<e0ae5108>] __67a1d0+0x150/0x1d0 [dvb]
[  538.566000] [<8042e3d0>] misc_open+0x168/0x1e0
[  538.570000] [<801363ac>] chrdev_open+0xc8/0x194
[  538.575000] [<8012dab8>] do_dentry_open.isra.2+0x26c/0x468
[  538.580000] [<801415c4>] path_openat+0x2b4/0x10e8
[  538.585000] [<801436d0>] do_filp_open+0xc8/0x120
[  538.590000] [<8012f34c>] do_sys_open+0x194/0x274
[  538.594000] [<80015c14>] syscall_common+0x34/0x58
[  538.599000]
[  538.600000]
[  538.600000] Code: 26140610  8ea20000  02802021 <8c450008> 0240f809  2406007c  3c02e0ae  2442468c  ae020634
[  538.611000] ---[ end trace e2cb657a6e4f1668 ]---
root@et10000:~#
```

atvcaptain, 2017-03-16:
sure wait new update coming use old image and dont update

original-birdman, 2017-03-27:
> sure wait new update coming use old image and dont update

When? _You_ are still denying these Xtrend models a working kernel when one is available.

ccs-ccs, 2017-04-01:
Is it really over 2 months since I first raised this issue?

original-birdman, 2017-04-01:
Is it _really_ nearly a month since I posted a fix (to revert to an kernel that actually works) for OpenVix (only), that was rejected by non-OpenViX people?

atvcaptain, 2017-04-02:
internal we test new version  with fix , and new features

ccs-ccs, 2017-04-02:
> @atvcaptain; internal we test new version with fix , and new features

That's good to hear, the "new features" sound intriguing.

Thanks again.

original-birdman, 2017-04-02:
> internal we test new version with fix , and new features

Meanwhile the _rest_ of us have to live with a broken kernel, even though there was a perfectly good fix presented which you rejected?
And since you didn't see anything wrong with the 4.8.3 kernel (when there manifestly was), how confident should we be that you spot any similar problems with this new version?

atvcaptain, 2017-04-02:
https://youtu.be/nfrYpAPFQlw

original-birdman, 2017-04-02:
OK - it looks promising. Let's hope it fixes things for all USB tuners,. and USB Wifi.

RobvanderDoes, 2017-04-02:
I fail to see why ViX isn't testing this. After all ViX was (at least as far as I know) the only image that had issues with the current unofficial drivers.

atvcaptain, 2017-04-13:
https://github.com/oe-alliance/oe-alliance-core/commit/51ee8486c183b8e34be6a330f4bff92a68f5ed75

now fixed

closed 2017-04-13 by atvcaptain

original-birdman, 2017-04-14:
Not fixed for the PCTV 292e triplestick, as it no longer loads any firmware, so now does nothing.

Commented at:https://github.com/oe-alliance/oe-alliance-core/commit/51ee8486c183b8e34be6a330f4bff92a68f5ed75

atvcaptain, 2017-04-14:
fix it and add correct firmware , that is the problem ?

atvcaptain, 2017-04-14:
https://www.linuxtv.org/wiki/index.php/PCTV_Systems_tripleStick_T2_(292e)

original-birdman, 2017-04-14:
I do have the correct firmware in place. It was all working with the 4.0.1 kernel. It crashed with the 4.8.3 kernel, and with the 4.10.6 kernel is doesn't download the firmware to the stick.
My fix was to revert to the WORKING 4.0.1 kernel, but _you_ won't let OpenViX do that.

xtrendsupport, 2017-04-14:
Revert a kernel update is not a fix... 
Can you tell me what the manufacture PCTV say about this issue.
  
Please, if possible can you send a new dmesg and lsmod log with the new kernel. 
Maybe stupid question but did you install kernel-module-tveeprom and all other necessary modules

atvcaptain, 2017-04-14:
you have not read the info linux tv org, driver and firmware must use the correct version

> Please note that last version of the driver is needed as there is a strong symbiosis between drivers and firmware.

original-birdman, 2017-04-14:
> Please, if possible can you send a new dmesg and lsmod log with the new kernel.

For dmesg see the first comment on:
https://github.com/oe-alliance/oe-alliance-core/commit/51ee8486c183b8e34be6a330f4bff92a68f5ed75#commitcomment-21762957

lsmod is:
```
Module                  Size  Used by
ipv6                  418307  33
si2157                  5077  1
si2168                  7064  1
em28xx_dvb             25315  0
em28xx                 67117  1 em28xx_dvb
tveeprom               10969  1 em28xx
xfs                   779193  0
libcrc32c                890  1 xfs
dvb                  7392270  6
modloader2             33169  1 dvb
modloader              22463  0
tpm                     6717  3 modloader,dvb,modloader2
```

original-birdman, 2017-04-14:
> Revert a kernel update is not a fix...

It isn't a fix for the kernel, but **it is a fix** for a broken OpenViX image.

original-birdman, 2017-04-14:
> you have not read the info linux tv org, driver and firmware must use the correct version

I have read it. I have tried different versions of the firmware. **You** haven't read that I posted a link to where you can find 5 different versions.

xtrendsupport, 2017-04-14:
Your (stripped?) log given no errors. Actually the device is initialize in a right way.
em28xx 1-1:1.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...

The missing part is that enigma2 should make a Vtuner device so that's accessible as a nim device.

original-birdman, 2017-04-14:
There are no errors. It just never gets round to downloading the firmware file.

> The missing part is that enigma2 should make a Vtuner device so that's accessible as a nim device.
As I _keep_ saying - all of this was working fine with the 4.0.1 kernel.

Huevos, 2017-04-15:
> The missing part is that enigma2 should make a Vtuner device so that's accessible as a nim device

How can it be a fault in E2? It works fine on every other brand of STB and also on the xtrend when using the older drivers.

original-birdman, 2017-04-15:
>     The missing part is that enigma2 should make a Vtuner device so that's accessible as a nim device

OpenViX 5.0.011 (release). uses kernel 4.8.3. When I start-up enigma2 with the triplestick in place, the firmware is downloaded to it and the dvb module crashes.

OpenVix 5.0.012.001 (dev). uses kernel 4.10.6. When I start-up enigma2 with the triplestick in place the firmware is **not** downloaded. The tuner is **not** visible to ViX.

The only difference between the enigma2 source code for those two versions on git is a few differences in some *.py files, non of which has anything to do with discovering tuners.

atvcaptain, 2017-04-15:
ask steff send me a usb stick and i fix it for you

ccs-ccs, 2017-04-15:
@original-birdman Any chance you could build an ET10k 4.10.6 for me to try out the SundTek usb tuner?
You could pop it in OpenVix/debug/CCS as before.
Cheers
ccs

original-birdman, 2017-04-15:
> ask steff send me a usb stick and i fix it for you

I doubt that he has one. He doesn't sell them.

PS; Just noticed one thing about the 4.8.3 kernel crashes.

The "standard" 3 tuners show this:
```
Jan  1 01:00:16 et8000 user.info kernel: [    9.455000] platform dvb0.0: DVB: registering adapter 0 frontend 0 (DVB-T/T2/C)...
Jan  1 01:00:16 et8000 user.info kernel: [    9.471000] platform dvb0.0: DVB: registering adapter 0 frontend 1 (DVB-T/T2/C)...
Jan  1 01:00:16 et8000 user.info kernel: [    9.481000] platform dvb0.0: DVB: registering adapter 0 frontend 2 (DVB-T/T2/C)...
```
then the USB tuner shows as this (first plugged in)
Jan 24 16:43:01 et8000 user.info kernel: [ 6271.396000] usb 1-1: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
```
then this, as enigma2 stars up
```
Jan 24 16:43:43 et8000 user.info kernel: [ 6311.381000] platform dvb0.0: DVB: registering adapter 0 frontend 3 (vtuner)...
```
So although the tuner was adapter 1, frontend 0, the xtrend dvb module was trying to do something with the non-existent adapter 0 frontend 3.

With the 4.10.6 dvb module it doesn't see this USB tuner at all.

original-birdman, 2017-04-15:
Looking through the code I see that enigma2 looks as the contents of `/proc/bus/nim_sockets` to get the list of available tuners.
With the 4.0.1 kernel, as the GUI starts up this contains an entry for the USB stick:
```
NIM Socket 3:
        Type: DVB-T2
        Name: PCTV 292e
        Has_Outputs: no
        Internally_Connectable: 0
        Frontend_Device: 3
        I2C_Device: -1
```
With the 4.10.6 kernel it never does.

I suspect that the contents of this are determined by the kernel dvb module?

original-birdman, 2017-04-15:
Looking through the code, enigma2 will run with the tuners it finds in `/proc/bus/nim_sockets`.
It also look at /dev/dvb/adaptor0, adaptor1, _etc._..

When I insert the PCTV stick into a system with 1 4.0.1 kernel, /dev/dvb/adaptor1 appears.
When I insert it into a 4.10.6 kernel this does not appear. This is why enigma2 isn't seeing it. And my suspicion is that this device file is managed by the Xtrend kernel dvb module, which is why I reckon it's still broken - just in a different way.
[Perhaps they decided to "fix" the crash by ignoring the device completely?]

original-birdman, 2017-04-15:
OK. I can confirm that the USB stick is actually working with the 4.10.6 kernel, but is is **NOT** being seen by enigma2.

If I copy the files for dvdv6-scan form a Debian mips system onto the box I can run a scan - this works. The scan is on adaptor 1, frontend 0.  teh firmware downlod works OK.

But the `/proc/bus/nim_sockets` file (which is what enigma2 uses to get the list of adaptors to use( never contains any info about the USB tuner.   The sysfs file, and `/dev/usb` files, do all get created as expected when I plug in the stick.

So the problem appears to be with the `/proc/bus/nim_sockets` file, which I presume is handled by the Xtrend kernel dvb module(?).

ccs-ccs, 2017-04-16:
I've now been able to test the SundTek  T2 usb tuner on 4.10.6 and can confirm that it works fine.

Here's my nim_sockets file.

What's odd is that sockets 2 and 3 have the same I2C_Device number.
 I wonder if this is why I can't record 4 different mux's using hybrid dvb-c/t/**t2** tuners?
All 4 work individually, but only 3 out 4 when all 4 are needed.
 Tuners C and D certainly share something, because Tuner C doesn't work when I remove the feed to tuner D.

```
root@et10000:~# cat  /proc/bus/nim_sockets
NIM Socket 0:
        Type: DVB-T2
        Name: Si2169
        Has_Outputs: yes
        Frontend_Device: 0
        I2C_Device: 2
NIM Socket 1:
        Type: DVB-T2
        Name: Si2169
        Has_Outputs: yes
        Frontend_Device: 1
        I2C_Device: 3
NIM Socket 2:
        Type: DVB-T2
        Name: Si2169
        Has_Outputs: yes
        Frontend_Device: 2
        I2C_Device: 4
NIM Socket 3:
        Type: DVB-T2
        Name: Si2169
        Has_Outputs: yes
        Frontend_Device: 3
        I2C_Device: 4
NIM Socket 4:
        Type: DVB-C
        Name: Sundtek DVB-C (III) (1/0) (USB)
        Has_Outputs: no
        Frontend_Device: 4
        I2C_Device: -1
root@et10000:~#
```

original-birdman, 2017-04-17:
> ask steff send me a usb stick and i fix it for you

I can see what he problem is!!!  (But I don't know the solution).

Wandering through the OpenViX source I found that it is looking for an ep_00 entry in `/sys/class/dcvb/dvb1.frontend0/device` to mark this as a USB tuner. So I had a look at the contents of this directory when running with the (working) 4.0.1 kernel and the (not working) 4.10.6 kernel.  The contents are quite different (I do have both listings). The former contains an ep_00, while the latter contains an ep_84.

Fortunately, the script I used to generate the listing (and the content of each file) started with running `/bin/pwd` (to get the physical location) and there is a (_very_) significant difference.

4.0.1 has: `/sys/devices/platform/ehci-brcm.0/usb1/1-1`
4.10.6 has: `/sys/devices/platform/ehci-brcm.0/usb1/1-1/1-1:1.0`

_i.e._ 4.10.6 is one level further down. And if I look at that one level higher it contains the same info as does 4.0.1.

So the problem would appear to be that whatever is setting up the symlinks in the sysfs is getting it wrong!

There are two symlinks along the way to `/sys/class/dcvb/dvb1.frontend0/device`, so I'll need to re-flash the 4.0.1 image to find out which the right ones are (and hence which the wrong one is in 4.10.6.

original-birdman, 2017-04-17:
As a result of this symlink oddity the `dvb` directory, and also the `i2c-7` and `i2c-8` ones, end up one level further down in the 4.10.6 kernel sysfs file-system. The fact that the `dvb` one is one level further down results in its device symlink pointing to the wrong location and enigma2 not realizing that it is a USB tuner.

atvcaptain, 2017-04-19:
https://youtu.be/0VyBUorxRhs

original-birdman, 2017-04-19:
Thanks, but it sill doesn't work for me on OpenVix.
The enigma2-plugin-drivers-ct2-dvb-usb-pctv292e package just adds kernel-module-em28xx-rc beyond what I already have, so I installed that (and removed kernel-module-si21xx, which was never used anyway).
boot up still shows this:
```
[root-et8000]: grep DVB /var/log/messages 
Jan  1 01:00:15 et8000 user.info kernel: [    7.923000] dvbdev: DVB: registering new adapter (dvb0)
Jan  1 01:00:15 et8000 user.info kernel: [    7.929000] platform dvb0.0: DVB: registering adapter 0 frontend 0 (DVB-T/T2/C)...
Jan  1 01:00:15 et8000 user.info kernel: [    7.929000] platform dvb0.0: DVB: registering adapter 0 frontend 1 (DVB-T/T2/C)...
Jan  1 01:00:15 et8000 user.info kernel: [    7.929000] platform dvb0.0: DVB: registering adapter 0 frontend 2 (DVB-T/T2/C)...
Jan  1 01:00:15 et8000 user.err kernel: [    9.514000] em28xx 1-1:1.0: DVB interface 0 found: isoc
Jan  1 01:00:15 et8000 user.info kernel: [   11.500000] em28xx 1-1:1.0: Binding DVB extension
Jan  1 01:00:15 et8000 user.info kernel: [   11.558000] dvbdev: DVB: registering new adapter (1-1:1.0)
Jan  1 01:00:15 et8000 user.info kernel: [   11.558000] em28xx 1-1:1.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
Jan  1 01:00:15 et8000 user.info kernel: [   11.559000] em28xx 1-1:1.0: DVB extension successfully initialized
```
with em28xx registering the device as 1-1:1.0 (i.e. one level below where it "should" be).

With the 4.0.1 kernel this looked like:
```[root-et8000]: grep DVB startup-messages.4.0.1 
Jan  1 01:00:16 et8000 user.info kernel: DVB: registering new adapter (dvb0)
Jan  1 01:00:16 et8000 user.info kernel: platform dvb0.0: DVB: registering adapter 0 frontend 0 (DVB-T/T2/C)...
Jan  1 01:00:16 et8000 user.info kernel: platform dvb0.0: DVB: registering adapter 0 frontend 1 (DVB-T/T2/C)...
Jan  1 01:00:16 et8000 user.info kernel: platform dvb0.0: DVB: registering adapter 0 frontend 2 (DVB-T/T2/C)...
Jan  1 01:00:16 et8000 user.info kernel: em28xx: DVB interface 0 found: isoc
Jan  1 01:00:16 et8000 user.info kernel: em28178 #0: Binding DVB extension
Jan  1 01:00:16 et8000 user.info kernel: DVB: registering new adapter (em28178 #0)
Jan  1 01:00:16 et8000 user.info kernel: usb 1-1: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
Jan  1 01:00:16 et8000 user.info kernel: em28178 #0: DVB extension successfully initialized
Apr 19 02:26:10 et8000 user.info kernel: platform dvb0.0: DVB: registering adapter 0 frontend 3 (vtuner)...
```
with the em28xx register being different, and then "platform" picking it up for the vtuner.

What does the OpenATV message file show for this?

atvcaptain, 2017-04-19:
check atv e2 git and update vix e2 fix your issue

original-birdman, 2017-04-19:
Well, that makes the tuner show up. A recording started just after I tested that, so I'll have to wait to check that the tuner is usable (it's certainly configurable, as I had to switch it from the default DVB-C to DVB-T2).
The code, though, strikes me as a horrible kludge. I have a feeling that it should be check for something specific not "is endpoint 0 there - if not do we happen to be somewhere else with an endpoint 84, so there must be an endpoint 0 above it....".

But many thanks anyway.

original-birdman, 2017-04-19:
I can confirm that it **is** now working.

(It can't get a signal from the 2 low-power muxes broadcast to me - something I've only noticed as a result of this testing, but it's fine with the other ones).

original-birdman, 2017-04-19:
> The code, though, strikes me as a horrible kludge.

My gut feeling is that you should read the symlink at `/sys/class/dvb/dvb%d.frontend%d/device/subsystem`
and check that it ends with `/bus/usb`.

original-birdman, 2017-04-20:
https://github.com/OpenViX/enigma2/pull/160

```
bool eDVBAdapterLinux::isusb(int nr)
{
        bool res = false;
        char subsystem_name[256], link_name[256];
        snprintf(subsystem_name, sizeof(subsystem_name),
                "/sys/class/dvb/dvb%d.frontend0/device/subsystem", nr);
        ssize_t link_len = ::readlink(subsystem_name, link_name, sizeof(link_name));
        if (link_len >= 4)  //* All 4's here are the length of "/usb"
                res = (strncmp("/usb", link_name+link_len-4, 4) == 0);
        return res;;
}
```