oe-alliance/oe-alliance-core #190: OpenATV 6.0 vu+ultimo 4k

CHazz, 2017-04-10:
I'm sorry that I write here .. can not speak German as millions of users .. openATV their support is only in German and has no bugtacker etc .. sorry again .. ultimo 4k everything works perfectly. except hdmi input .. preview works but can not record in 1080p thank you for the correction .. in VTI this works .. but I hate VTI :)

atvcaptain, 2017-04-11:
input works in 1080P with vti ?  this info is new we have only info max 720p possible

looks vti have non public infos from vu we dont have it

Schimmelreiter, 2017-04-11:
There is an English section in the OpenATV forum.
It's just somewhat silent as OpenATV is close to perfection ;)

Ev0-BH, 2017-04-11:
Last time I checked VTi, it was still only 720p recording

closed 2017-04-11 by atvcaptain