oe-alliance/oe-alliance-core #203: CONFIG_INPUT_ATI_REMOTE2

nationat, 2017-06-22:
Hi,
please add CONFIG_INPUT_ATI_REMOTE2=y in next Kernel Release. 

It supports thousands of remote controlls (e.g. Medion, ATI Remote Wonder, many X10 RF RC's)

Thanx a lot.

nationat, 2017-07-03:
**Sorry, what i rearly need is ati_remote with rc-core:**
 modinfo ati_remote
filename:       /lib/modules/4.4.0-28-generic/kernel/drivers/media/rc/ati_remote.ko
license:        GPL
description:    ATI/X10 RF USB Remote Control
author:         Torrey Hoffman <thoffman@arnor.net>
srcversion:     D1EB4EDCFB4B6FD2C903059
alias:          usb:v0BC7p0008d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BC7p0006d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BC7p0005d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BC7p0004d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BC7p0003d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BC7p0002d*dc*dsc*dp*ic*isc*ip*in*
depends:        rc-core
intree:         Y
vermagic:       4.4.0-28-generic SMP mod_unload modversions
parm:           channel_mask:Bitmask of remote control channels to ignore (ulong)
parm:           debug:Enable extra debug messages and information (int)
parm:           repeat_filter:Repeat filter time, default = 60 msec (int)
parm:           repeat_delay:Delay before sending repeats, default = 500 msec (int)
parm:           mouse:Enable mouse device, default = yes (bool)

**but the ati_remote2 is also beautiful (for other people with rf remote):**
modinfo ati_remote2
filename:       /lib/modules/4.4.0-28-generic/kernel/drivers/input/misc/ati_remote2.ko
license:        GPL
author:         Ville Syrjala <syrjala@sci.fi>
version:        0.3
description:    ATI/Philips USB RF remote driver
srcversion:     01EAEA89947A935915FE5EE
alias:          usb:v0471p0602d*dc*dsc*dp*ic*isc*ip*in*
depends:
intree:         Y
vermagic:       4.4.0-28-generic SMP mod_unload modversions
parm:           channel_mask:Bitmask of channels to accept <15:Channel16>...<1:Channel2><0:Channel1> (channel_mask)
parm:           mode_mask:Bitmask of modes to accept <4:PC><3:AUX4><2:AUX3><1:AUX2><0:AUX1> (mode_mask)

atvcaptain, 2017-07-03:
https://github.com/oe-alliance/oe-alliance-core/commit/c04f12c0250915afaa90d36f9a9e0fd2ac900b29

closed 2017-07-03 by atvcaptain

nationat, 2017-07-04:
Hi atvcaptain,

thanx for adding ati_remote2. But please add also the module ati_remote (without the "2", it is for different Remotes!)
The rc-core module is present in defconfig:

Media drivers

CONFIG_RC_CORE=y
CONFIG_RC_MAP=y
**CONFIG_RC_ATI_REMOTE=m**