oe-alliance/oe-alliance-core #204: Plz update the sabnzbd for oatv

heisert, 2017-06-23:
The last update was a very long time ago. Atm its not working very well because the unrar version is outdated. Many files cant be extracted while in Version 5 the password form got a change. 
It's not working well on mips/arm box versions. 

I hope we see a update (maybe to 2.1) or only to a newer unrar 5+ version. It would help a lot. 

Thanks

atvcaptain, 2017-06-30:
done sabnzbd to 1.10 and unrar to 5.5.5

closed 2017-06-30 by atvcaptain

heisert, 2017-07-01:
Thanks for updating but It's not working with newest sabzbd. Cant start the Service on the network/service

atvcaptain, 2017-07-01:
please check this
opkg update
opkg install python-multiprocessing

heisert, 2017-07-02:
Works got only this errors 

WARNING vor 3 Minuten SABYenc module... NOT found! Expecting v3.0.2 - https://sabnzbd.org/sabyenc

WARNING vor 3 Minuten SABnzbd was started with encoding ANSI_X3.4-1968, this should be UTF-8. Expect problems with Unicoded file and directory names in downloads.

Speed up repairs by installing multicore Par2, it is available for many platforms. https://sabnzbd.org/wiki/installation/multicore-par2

arn354, 2017-07-02:
root@gbquad4k:~# opkg list |grep python-sabyenc
python-sabyenc - 3.0.2-r0 - sabyenc - yEnc Module for Python modified for SABnzbd
python-sabyenc-dbg - 3.0.2-r0 - sabyenc - yEnc Module for Python modified for SABnzbd - Debugging files
root@gbquad4k:~#

successfully build the package, i add it - we will see, i do not use SABnzbd

heisert, 2017-07-03:
Cool but how to install?

heisert, 2017-07-03:
Could this help for the utf-8 encoding warning? 
https://github.com/sabnzbd/sabnzbd/issues/831

heisert, 2017-07-07:
opkg install python-sabyenc
Couldn't find anything to satisfy 'python-sabyenc'.
Unknown package 'python-sabyenc'.
Collected errors:
 * opkg_solver_install: Cannot install package python-sabyenc.

arn354, 2017-07-07:
the bb is not yet pushed to build

atvcaptain, 2017-07-07:
done

heisert, 2017-07-09:
Seems its working! Warning is gone. _Thanks_ arn354. 

Only 1 warning left:

WARNING vor 2 Minuten SABnzbd was started with encoding ANSI_X3.4-1968, this should be UTF-8. Expect problems with Unicoded file and directory names in downloads.

atvcaptain, 2017-07-09:
i dont use this, send info that must be change