oe-alliance/oe-alliance-core #211: CONFIG_RC_ATI_REMOTE

nationat, 2017-07-06:
[](relates to: https://github.com/oe-alliance/oe-alliance-core/commit/c04f12c0250915afaa90d36f9a9e0fd2ac900b29)

thanx for adding ati_remote2. But please add also the module ati_remote (without the "2", it is for different Remotes!)
The rc-core module is present in defconfig:

Media drivers

CONFIG_RC_CORE=y
CONFIG_RC_MAP=y
**CONFIG_RC_ATI_REMOTE=m**

modinfo ati_remote
filename: /lib/modules/4.4.0-28-generic/kernel/drivers/media/rc/ati_remote.ko
license: GPL
description: ATI/X10 RF USB Remote Control
author: Torrey Hoffman thoffman@arnor.net
srcversion: D1EB4EDCFB4B6FD2C903059
alias: usb:v0BC7p0008ddcdscdpiciscipin
alias: usb:v0BC7p0006ddcdscdpiciscipin
alias: usb:v0BC7p0005ddcdscdpiciscipin
alias: usb:v0BC7p0004ddcdscdpiciscipin
alias: usb:v0BC7p0003ddcdscdpiciscipin
alias: usb:v0BC7p0002ddcdscdpiciscipin
depends: rc-core
intree: Y
vermagic: 4.4.0-28-generic SMP mod_unload modversions
parm: channel_mask:Bitmask of remote control channels to ignore (ulong)
parm: debug:Enable extra debug messages and information (int)
parm: repeat_filter:Repeat filter time, default = 60 msec (int)
parm: repeat_delay:Delay before sending repeats, default = 500 msec (int)
parm: mouse:Enable mouse device, default = yes (bool)

nationat, 2017-07-11:
Tested. For me it works with:

Media drivers

CONFIG_RC_CORE=y
CONFIG_RC_MAP=y
**CONFIG_RC_DEVICES=y
CONFIG_RC_ATI_REMOTE=m**

atvcaptain, 2017-09-30:
add now

closed 2017-09-30 by atvcaptain