oe-alliance/oe-alliance-core #221: Support for TP-LINK 722N v2 WIFI USB adapters

devlo, 2017-08-16:
Hi,
Any plans to support updated version of new TP-LINK 722N v2 wifi usb adapters?

https://wikidevi.com/wiki/TP-LINK_TL-WN722N_v2

Becasue they are not working on openatv 6.0/6.1 nightly.

They were added in april this year to rtl8188eu driver:

https://github.com/lwfinger/rtl8188eu/issues/206

atvcaptain, 2017-09-30:
include

closed 2017-09-30 by atvcaptain

mesagac, 2018-07-09:
Hi,
It seems driver is included in kernel (openatv 6.1) but no response. No wlan connection menu appears after plugging and then restarting the box.

atvcaptain, 2018-07-09:
you must check dmesg log after insert wlan stick

mesagac, 2018-07-11:
[    0.009000] bcm40nm_pm_moca_disable 00
[    0.009000] smp: Bringing up secondary CPUs ...
[    0.010000] SMP: Booting CPU1...
[    0.010000] Primary instruction cache 32kB, physically tagged, 4-way, linesize 64 bytes.
[    0.010000] Primary data cache 32kB, 4-way, linesize 32 bytes.
[    0.010000] MIPS secondary cache 256kB, 8-way, linesize 128 bytes.
[    0.010000] CPU1 revision is: 00025a11 (Broadcom BMIPS5000)
[    0.010000] FPU revision is: 00130001
[    0.019000] SMP: CPU1 is running
[    0.019000] smp: Brought up 1 node, 2 CPUs
[    0.019000] devtmpfs: initialized
[    0.022000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275000 ns
[    0.022000] futex hash table entries: 512 (order: 4, 65536 bytes)
[    0.022000] NET: Registered protocol family 16
[    0.036000] USB0: power enable is active high; overcurrent is active low
[    0.036000] USB1: power enable is active high; overcurrent is active low
[    0.045000] SCSI subsystem initialized
[    0.045000] libata version 3.00 loaded.
[    0.045000] usbcore: registered new interface driver usbfs
[    0.045000] usbcore: registered new interface driver hub
[    0.045000] usbcore: registered new device driver usb
[    0.046000] Linux cec interface: v0.10
[    0.046000] Linux video capture interface: v2.00
[    0.046000] Advanced Linux Sound Architecture Driver Initialized.
[    0.046000] Bluetooth: Core ver 2.22
[    0.046000] NET: Registered protocol family 31
[    0.046000] Bluetooth: HCI device and connection manager initialized
[    0.046000] Bluetooth: HCI socket layer initialized
[    0.046000] Bluetooth: L2CAP socket layer initialized
[    0.046000] Bluetooth: SCO socket layer initialized
[    0.047000] clocksource: Switched to clocksource wktmr
[    0.047000] FS-Cache: Loaded
[    0.054000] random: fast init done
[    0.056000] NET: Registered protocol family 2
[    0.057000] TCP established hash table entries: 8192 (order: 3, 32768 bytes)
[    0.057000] TCP bind hash table entries: 8192 (order: 4, 65536 bytes)
[    0.057000] TCP: Hash tables configured (established 8192 bind 8192)
[    0.057000] UDP hash table entries: 512 (order: 2, 16384 bytes)
[    0.057000] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes)
[    0.057000] NET: Registered protocol family 1
[    0.058000] RPC: Registered named UNIX socket transport module.
[    0.058000] RPC: Registered udp transport module.
[    0.058000] RPC: Registered tcp transport module.
[    0.058000] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    0.059000] workingset: timestamp_bits=14 max_order=17 bucket_order=3
[    0.059000] FS-Cache: Netfs 'nfs' registered for caching
[    0.060000] NFS: Registering the id_resolver key type
[    0.060000] Key type id_resolver registered
[    0.060000] Key type id_legacy registered
[    0.060000] nfs4filelayout_init: NFSv4 File Layout Driver Registering...
[    0.060000] Installing knfsd (copyright (C) 1996 okir@monad.swb.de).
[    0.060000] FS-Cache: Netfs 'cifs' registered for caching
[    0.060000] Key type cifs.spnego registered
[    0.060000] fuse init (API version 7.26)
[    0.066000] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 253)
[    0.066000] io scheduler noop registered
[    0.066000] io scheduler cfq registered (default)
[    0.067000] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    0.068000] serial8250.0: ttyS0 at MMIO 0x10406900 (irq = 65, base_baud = 5062500) is a 16550A
[    0.068000] serial8250.0: ttyS1 at MMIO 0x10406940 (irq = 66, base_baud = 5062500) is a 16550A
[    0.069000] serial8250.0: ttyS2 at MMIO 0x10406980 (irq = 67, base_baud = 5062500) is a 16550A
[    0.076000] loop: module loaded
[    0.076000] ahci ahci.0: SSS flag set, parallel bus scan disabled
[    0.086000] ahci ahci.0: AHCI 0001.0300 32 slots 3 ports 6 Gbps 0x3 impl platform mode
[    0.086000] ahci ahci.0: flags: ncq sntf stag clo pmp slum part 
[    0.091000] scsi host0: ahci
[    0.092000] scsi host1: ahci
[    0.092000] scsi host2: ahci
[    0.092000] ata1: FORCE: PHY spd limit set to 3.0Gbps
[    0.092000] ata1: SATA max UDMA/133 mmio [mem 0x10181000-0x10190fff] port 0x100 irq 41
[... 119 lines cut ...]
[    1.654000] UBIFS (ubi0:0): UBIFS: mounted UBI device 0, volume 0, name "rootfs"
[    1.654000] UBIFS (ubi0:0): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[    1.654000] UBIFS (ubi0:0): FS size: 496603136 bytes (473 MiB, 3911 LEBs), journal size 9023488 bytes (8 MiB, 72 LEBs)
[    1.654000] UBIFS (ubi0:0): reserved for root: 0 bytes (0 KiB)
[    1.654000] UBIFS (ubi0:0): media format: w4/r0 (latest is w5/r0), UUID C81EF1DE-C82F-43E8-9788-B021D16FD85F, small LPT model
[    1.655000] VFS: Mounted root (ubifs filesystem) on device 0:12.
[    1.657000] devtmpfs: mounted
[    1.658000] Freeing unused kernel memory: 276K
[    1.658000] This architecture does not have kernel memory protection.
[    1.817000] random: crng init done
[    2.554000] formuler1_1: module_layout: kernel tainted.
[    2.554000] Disabling lock debugging due to kernel taint
[    2.554000] formuler1_1: loading out-of-tree module taints kernel.
[    2.558000] usbcore: registered new interface driver tpm
[    7.943000] input: dreambox remote control (native) as /devices/virtual/input/input0
[    7.944000] input: dreambox advanced remote control (native) as /devices/virtual/input/input1
[    8.018000] dvbdev: DVB: registering new adapter (dvb0)
[... cut ...] VB-S2)...
[    8.029000] platform dvb0.0: DVB: registering adapter 0 frontend 1 (DVB-S2)...
[    8.320000] input: front panel as /devices/virtual/input/input2
[    8.392000] Registered IR keymap rc-cec
[    8.393000] input: RC for hdmi_cec as /devices/platform/hdmi.0/rc/rc0/input3
[    8.393000] rc rc0: RC for hdmi_cec as /devices/platform/hdmi.0/rc/rc0
[    8.739000] SGI XFS with security attributes, realtime, no debug enabled
[    8.795000] boxmodel: module license 'proprietary' taints kernel.

atvcaptain, 2018-07-11:
cant see your wlan

boot box and insert usb stick after boot, check dmesg log for changes last lines and send not full log

mesagac, 2018-07-16:
Thanks for quick reply,

After the boot I inserted   TP-LINK 722N v3 and get these log lines. 
"It seems there is no WLAN again"

 [    2.542000] usbcore: registered new interface driver tpm
[    7.937000] input: dreambox remote control (native) as /devices/virtual/input/input0
[    7.938000] input: dreambox advanced remote control (native) as /devices/virtual/input/input1
[    8.012000] dvbdev: DVB: registering new adapter (dvb0)
[    8.015000] platform dvb0.0: DVB: registering adapter 0 frontend 0 (DVB-S2)...
[    8.029000] platform dvb0.0: DVB: registering adapter 0 frontend 1 (DVB-S2)...
[    8.313000] input: front panel as /devices/virtual/input/input2
[    8.396000] Registered IR keymap rc-cec
[    8.397000] input: RC for hdmi_cec as /devices/platform/hdmi.0/rc/rc0/input3
[    8.397000] rc rc0: RC for hdmi_cec as /devices/platform/hdmi.0/rc/rc0
[    8.732000] SGI XFS with security attributes, realtime, no debug enabled
[    8.769000] boxmodel: module license 'proprietary' taints kernel.