oe-alliance/oe-alliance-core #245: Cron Timers

kiddac, 2017-11-21:
What ever amends you have done with cron timers has totally broken crons.

Cronie doesn't seem to run any crons at all now

My crons live in /etc/cron/crontabs
for example 
0 9 * * * /usr/script/onyx-scrape.sh
*/30 * * * * /usr/script/onyx-pick.sh

I have also tried them in cron.d all folders and all files have correct permissions. As do the scripts.

I tried uninstalling cronie and reinstalling busybox-cron but that then adds further problems. It then crashes Infopanel crontimers if you try and start or autostart.

If i go to timers menu cron timers, it asks to reinstall cronie (which doesn't work).

Can you please go back to the old working busybox-cron instead of this cronie version that no matter what i try I cannot get to work. 

** using a zgemma **

regards 

KiddaC

Schimmelreiter, 2017-11-22:
cronjobs are **never** edited directly, no matter what cron is used and no matter what Linux distro is in use, but what I guess you are trying to do, reading that you talk about file locations.
It simply does not matter where they reside, as long as you edit them correctly.

The helper tool "crontab" makes sure that you edit the correct crontabs file, no matter as which user you are logged in and no matter where the *ix distro you are running puts them.
It also makes sure that changes to the crontabs are automatically applied.

Edit cronjobs manually:
```crontab -e```

Listing currently defined cronjobs:
```crontab -l```

Create a new cronjob list using a pipe's output:
```somecommands | crontab -```

The above commands can be concatenated for more sophisticated tasks for scripting cronjob additions/removals:

Add a cronjob from within a script:
```(crontab -l 2>/dev/null; echo "*/5 * * * * /path/to/job -with args") | crontab -```
where ```*/5 * * * * /path/to/job -with args``` is the new job.

Remove a cronjob from within a script:
```crontab -l 2>/dev/null | grep -v "sh /usr/script/hiddenmount.sh" | crontab -```
where ```sh /usr/script/hiddenmount.sh``` only needs to be enough to make the cronjob to be removed unique.

Lots of info at once, but the short truth is, that most users will only need ```crontab -e``` (Or in this special case, use the E2 GUI to edit them).

BTW: All of the above was true for busybox-cron already, it's not new.

Most likely, as once as you use "crontab -e" you will notice that you have wildly edited random wrong files.

kiddac, 2017-11-22:
Thank you for your reply.

I have always manually inserted crons into crontab before for busybox-cron and they have always worked no problem.

I was having problems whether I manually inserted the cron file. Or whether I used the add cron timers via the enigma2 menus. I did also briefly experiment with contab -e last night. With no luck. 
To me it seems cronie is not running although it says it is. 

Where do the cron logs live to possible debug this. 

Also have you tried setting a cron up yourself with this new cronie and have you had success in it running. If so could you possibly give an example of setting a cron up with this new cronie. 

Cheers

Schimmelreiter, 2017-11-22:
No, I always check in absolutely untested stuff ...
Of course I have tested it.

There is absolutely nothing wrong with cronie.
In this video you can see a minutely cronjob echoing "Test" to /tmp/crontest and the file coming back every minute if you delete it.
You can also see the file not coming back once the cronjob gets deleted:

https://youtu.be/nGuiVt3mK60

Schimmelreiter, 2017-11-22:
And here you can see that the file edited with crontab -e is the same as the one that gets edited in E2 GUI:
[image]

kiddac, 2017-11-22:
Ok thanks, I will report back once i have chance to further test. 
Would running "opkg update" cause any problems with this. 
I need to do that to install the full wget.

Schimmelreiter, 2017-11-22:
I suggest you do
1. ```opkg --force-depends remove busybox-cron```
and
2. ```rm -rf /etc/cron*```
and then
3. perfom an online flash with (new) settings backup and restore in order to get a clean cron environment again (You probably have messed around with the system-wide crontabs file too).

```opkg update``` is in no way related to this all, so you are free to do your ```opkg update && opkg install wget``` before step 3.

kiddac, 2017-11-22:
Over the last few days, I have people email about this problem. All on zgemmas at the moment. 
zgemma 2s. Zgemma h2s. And a Zgemma H2H.

kiddac, 2017-11-22:
Schimmelreiter,

I have done some more testing tonight. It just doesn't work. 

I even installed nano editor and did the exact same steps as in your youtube video. No file gets created in tmp. 

This was was a brand new openatv 6.1 install. Nothing else installed other than a skin. 
I did ...
opkg update
opkg install wget
opkg install nano

They are the only things I did on a fresh install. 

ran crontab -e
* * * * * echo "Test" > /tmp/crontest
Wrote to file. File is created in crontabs root. Waited.. absolutely nothing.

Went into Cron Timers menu. Crons are started. Crons are Autostarting. 

I have also seen someone mention the same problem with crons not working on a new flash on a Edision Mini last night as well. 

I think this has suddenly come to light this week and not the end of October when it was brought in, as everyone is doing an update or a reflash to fix the virgin audio problem. 

Is there anything else you can suggest or any files you wish me to send to you to see what the problem is.

kiddac, 2017-11-22:
them stange characters above were actually typed as asterisks ;)

kiddac, 2017-11-22:
OK I have finally got this working. 
I tried doing the exact steps in your video even going as far as using nano editor.
The first time it didn't seem to work. 
So I tried it again. The second time it worked. Also kicking into action my other script. Which was failing before the second success.

Looking at my error messages I was getting this originally.
Nov 22 20:46:01 zgemmas2s cron.info crond[1060]: (root) BAD FILE MODE (/var/spool/cron/crontabs/root)

Do you know what was causing this BAD FILE MODE

cheers

Schimmelreiter, 2017-11-22:
Creation/editing of all config files on the box from outside (via ftp/Samba/NFS) includes the risk to create them with too high rights or wrong line endings.
And yes, for cron using crontab avoids this. 
I'll have a look at the package to have it include a sample file, if busybox-cron did.

kiddac, 2017-11-22:
ok thanks. 

I reflashed again. manually created a root file and changed the permissions to 600. This still never worked.
It only works if I use crontab -e. Then write out file. Then it works straight away. 

Do you know of anyway I can just manually input this as a text file. 
I need to make it as simple as possible for people who are not very technically minded for people who use this script in my skins. 

cheers

Schimmelreiter, 2017-11-22:
Try if
```flip -u /etc/cron/crontabs/root```
makes the manually created crontab work.

In this case you know the problem was that you used normal line endings rather than *ix line endings.

kiddac, 2017-11-22:
I will close this thread now. Thanks a lot.

I can't really test the above command now as I have got everything up and running smoothly.

Crons work with crontab -e
Crons seem to work via the menus.
My problem was my cron was running every 15 minutes. There is no way to manually enter this in the cron timer menu as you can only enter 2 digits. This is why I have always done the manual way.

It does seem to work if I set it via the menus. I had to wait an hour to see :). And once this has been setup I can then edit it via a text editor. which  then works fine.

As mentioned above busybox-cron used to have an empty root file in crontab folder. So I am assuming it was already setup. Where as cronie doesn't. This I believe was causing my problems from the way I was used to doing it.

And just as a last question. What exactly is the difference between busybox-cron and cronie and why the change to the new one.

Schimmelreiter, 2017-11-23:
cronie has all the features and busybox-cron only a limited subset.

E.g. you can define cronjobs to be performed at boot using @reboot or hourly using @hourly (And @daily, @weekly, @monthly, @yearly respectively):
```@reboot       echo -n "False" > /tmp/was_timer_wakeup.txt```

Optionally you can even save the hassle of defining a cron job and just copy/link your script into /etc/cron.hourly , /etc/cron.daily , ... to have it executed hourly, daily, monthly, yearly respectively.
Just make sure the filename of the script contains only letters and digits (7 bit ASCII = a-z and A-Z, no umlauts etc.), the only "special" chars allowed are minus and underscores.

closed 2017-11-23 by kiddac