oe-alliance/oe-alliance-core #263: Core dumps on Octagon SF4008

pbasista, 2018-01-04:
I would need to debug an issue with a crashing application on Octagon SF4008. However, according to this:
https://github.com/oe-alliance/oe-alliance-core/blob/1b686f0a6dfd2cabc79007157da0186c1036393a/meta-brands/meta-octagon/recipes-linux/linux-octagon-4.1.37/sf4008/defconfig#L605
it does not have core dumps enabled, which complicates the procedure.

Would it be possible to enable core dumps for sf4008? And perhaps use a newer kernel, if it is not too much extra work?

Thank you.

atvcaptain, 2018-01-04:
enable this and build you own image for sf4008

kernel and close drivers comes from octagon we can change it

closed 2018-01-04 by atvcaptain

pbasista, 2018-01-04:
Ok, so, can you or can you not change it?

I can of course build my own image but the point of this issue is to enable it globally for everyone. The reason is that someone else might face a similar situation and they might need to go through that same procedure again, which is an unnecessary complication.

The core dumps are enabled for SF5008. I see no reason why they should be disabled for SF4008.

arn354, 2018-01-04:
check if your self-compiled Image with that change still is flashable and boots. Then it could be applied.

atvcaptain, 2018-01-04:
please test it in your local build first , close drivers not build with this option so i dont know drivers works after we change this, so we dont enable it

pbasista, 2018-01-10:
I have tested the build using this branch:
https://github.com/oe-alliance/build-enviroment/tree/4.2
And it worked fine. There were no issues related to the added kernel parameter and the resulting kernel image is perfectly usable on the receiver. On top of that, the core dumps work.

So, in my opinion, they can be enabled.

atvcaptain, 2018-01-10:
thx for testing we enable it