oe-alliance/oe-alliance-core #277: [dinobot 4k] Can anyone confirm SNRdB and Bitrate are working?

budiarno, 2018-03-15:
Hi guys.

Can any one confirm wether SNRdB and Bitrate (with BitrateCalc converter) are just working on Dinobot 4K? I've tried the same widget on my skin, but those two don't work as on other box (for example on Spark Box). Here's my skin screenshot:

[image]

Signal Strength is always showing "percentage", but not "dB".
Video Bitrate and Audio Bitrate is always showing "0 kbit/s".

atvcaptain, 2018-03-15:
issue allready report to driver team, you must wait

budiarno, 2018-05-04:
Thank you, it works now on Dinobot4K. Just needed to rework FrontendInfo converter.

closed 2018-05-04 by budiarno

atvcaptain, 2018-05-04:
you have rework any info for this ?

budiarno, 2018-05-04:
I've only edited my FrontendInfo.py on my local receiver. Haven't pushed the changes to any Enigma2 github sources at the moment.