oe-alliance/oe-alliance-core #285: H9 Screen Refresh Issue

p17blo, 2018-03-22:
I originally opened an issue against the enigma2 open atv git about the screen refresh on the H9 image.  The issue was closed and I was informed it was a driver issue outside the scope of the ATV team - I was given references to the following.

https://github.com/openatv/enigma2/commit/8cdda3d31a0e3a0ba95e22afaaa745a366a48921
https://github.com/oe-alliance/oe-alliance-core/commit/7f5a95d9ba7e55381075a8c55e87268f9d706dbe

I've followed the nightly builds, but in testing the screen refresh is still incredibly slow.  Is this something that is actively being worked on?

atvcaptain, 2018-03-22:
yes hiaccel fix comes with one of the coming updates, driver team works non stop to fix this

atvcaptain, 2018-03-23:
fixed with image 24.3

closed 2018-03-23 by atvcaptain