oe-alliance/oe-alliance-core #289: odhcp6c process running when no inet6 in /etc/network/interfaces

ewaldc, 2018-03-28:
**Expected behavior**: 
no odhcp6c process runs when /etc/network/interfaces does not contain a reference to _inet6_ (e.g. iface eth0 inet dhcp versus iface eth0 inet6 dhcp)
**Actual behavior**:
odhcp6c process is running for a given network interface despite _inet6_ being omitted from /etc/network/interfaces
On systems with low memory/limited virtual memory, this is an extra process that should not be running by design.
**Environment**: VU+ Solo SEv2 running OpenATV 6.2

I have traced back the issue to line 6 in oe-alliance-core/meta-oe/recipes-connectivity/odhcp6c/odhcp6c/ifup:
`grep -Eq '^iface  *'${IFACE}'  *inet6  *' /etc/network/interfaces && exit 0`
A possible fix is:
`grep -Eq '^iface  *'${IFACE}'  *inet6  *' /etc/network/interfaces || exit 0`

Happy to submit a pull request if needed.

Thanks for all the great work!
Ewald

atvcaptain, 2018-08-04:
add https://github.com/oe-alliance/oe-alliance-core/commit/277ebad7faf4635f4aad277ea588dd8d5dc1f733

closed 2018-08-04 by atvcaptain

Schimmelreiter, 2018-08-04:
Configure the interface in /etc/network/interfaces like this:
```iface eth0 inet6 manual```
replacing "eth0" with whatever interface you use.

In Linux it is accepted behaviour that an interface unconfigured for IPv6 means "IPv6 enabled".

Ignore idiots recommending deletion of this or that, patching sysctl.conf or disabling IPv6 entirely (As parts of E2 use IPv6 on loopback internally).

Just freaking disable it on the physical interfaces eth0, eth1, wlan0, wlan1, ... or whatever you use ...
And maybe someone will finally include an option inside the network setup of E2.

For the beginning, switching between
```iface eth0 inet6 auto``` (IPv6 auto (SLAAC/DHCPv6))"
```iface eth0 inet6 dhcp``` (IPv6 DHCPv6))"
```iface eth0 inet6 manual``` ("IPv6 disabled/manual")
would be enough.

inet6 auto is the most common way, it will use SLAAC or DHCP as needed and work with 99.9% of all routers.
inet6 manual prepares the interface for manual configuration and unless you really do manually configure it, it means "disabled".

Schimmelreiter, 2018-08-04:
For a reference, see https://github.com/oe-alliance/oe-alliance-core/blob/4.1/meta-oe/recipes-core/busybox/busybox-1.24.1/0002-Extended-network-interfaces-support.patch

You can see that "inet6 auto" will configure the interface for SLAAC (autoconf) and will start odhcp6c as RDNS-client (-Nnone = Don't try to get address via DHCP).
"inet6 dhcp" will configure the interface for SLAAC (autoconf) **and** dhcp (starting odhcp6c as DHCPv6 client with -Ntry).

And in contrast "inet6 manual" disables autoconf entirely and does **not** start odhcp6c at all.