oe-alliance/oe-alliance-core #300: OpenWebif new Design

rubix86, 2018-05-07:
Does OpenWebif new design work perfectly with OE-Alliance images or not? because it's does not work with me
[image]

arn354, 2018-05-13:
[image]

rubix86, 2018-05-13:
I use latest OpenBlackHole 4.1.028 image
Is openbh source from same source of OE-Alliance or not? 
when i use new desgin I got this
[image]
other images like OpenATV are ok and even VTi is ok so problem not from browser

arn354, 2018-05-14:
you should ask the image team to check and fix in their forums, not sure if they respond here.

AbuBaniaz, 2018-06-17:
You have to install the extra themes and then restart

Menu > Plugins > Extensions > Openwebif-themes