oe-alliance/oe-alliance-core #301: Strange colours on front panel LCD (GigaBlue Quad Plus) since driver update (8e2ab69)

outrageuk, 2018-05-08:
Strange colours on front panel LCD (GigaBlue Quad Plus) since driver update.

https://www.world-of-satellite.com/showthread.php?59392-Strange-colours-on-front-panel-after-latest-driver-update

atvcaptain, 2018-05-08:
vix team must change e2 code, no issue: <About LCD bit order>
 - Remove workaround code "color bit order" for above models in lcd.cpp or remove LCD_COLOR_BITORDER_RGB565 in configure.ac

ViX-Sicilian, 2018-05-08:
Theres nothing in our configure.ac for this.

Do we need these commits?

https://github.com/openatv/enigma2/commit/b0e17f7d6499bfad4f5fc0a55d16737988550b70#diff-295386107bd1763beba0be67031b8fa1

Anything else we need?

ViX-Sicilian, 2018-05-08:
That code is still in OpenATV lcd.cpp, so can anyone please advise exactly what we need to cherry pick.

betacentauri, 2018-05-08:
Afaik all gigablue boxes don't need the workaround any longer. Workaround was deactivated here:
https://github.com/openatv/enigma2/commit/1ba3e5601decb8b1b4eb0a9ae6d3ac94d7fcbab0#diff-67e997bcfdac55191033d57a16d1408a
https://github.com/openatv/enigma2/commit/14a9a1d2bdeb7160665b90d8116d0151df1e69ab#diff-67e997bcfdac55191033d57a16d1408a
ATV uses the workaround also for other boxes. It looks like you not. So I think you can remove the code.

betacentauri, 2018-05-08:
Sorry closed accidentially

outrageuk, 2022-07-07:
I seem to have this same problem again since updating for 5.2.

OpenVIX 6.2.006

Has this fix been removed in new builds for some reason?