oe-alliance/oe-alliance-core #309: ERROR: gettext-native-0.19.8.1-r0 do_compile: Function failed: do_compile - ubuntu 16.04.1

frankviana, 2018-05-15:
And about this error?

Initialising tasks: 100% |#######################################| Time: 0:00:52
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: gettext-native-0.19.8.1-r0 do_compile: oe_runmake failed
ERROR: gettext-native-0.19.8.1-r0 do_compile: Function failed: do_compile (log file is located at /home/ubuntu16/oealliance/build-enviroment/builds/openatv/release/alien5/tmp/work/x86_64-linux/gettext-native/0.19.8.1-r0/temp/log.do_compile.7645)
ERROR: Logfile of failure stored in: /home/ubuntu16/oealliance/build-enviroment/builds/openatv/release/alien5/tmp/work/x86_64-linux/gettext-native/0.19.8.1-r0/temp/log.do_compile.7645
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 1

atvcaptain, 2018-05-15:
check logs before this info, the falied not include

atvcaptain, 2018-05-15:
trash all, read here the build info and start new  https://github.com/openatv/enigma2/blob/6.1/README.md