oe-alliance/oe-alliance-core #324: How build only webkit?

maxbambi, 2018-06-28:
I have got an octagon sf4008 with openatv 6.2... There are problems to show hbbtv of RAI channels
Looking for errors in remote site (url http://www.replaytvmhp.rai.it/hbbtv/RaiPlay/index.html), 
i have seen that code uses a lot of svg objects...
In webkit recipes svg is disabled
`EXTRA_OECONF = "
	--enable-debug=no 
	--with-gtk=2.0 
	--disable-spellcheck 
	--enable-optimizations 
	--disable-channel-messaging 
	--disable-meter-tag 
	--enable-image-resizer 
	--disable-sandbox 
	--disable-xpath 
	--disable-xslt 
	--disable-svg 
	--disable-filters 
	--disable-svg-fonts 
	--disable-video 
	--disable-video-track 
	--with-target=directfb 
	--disable-jit 
	--enable-fast-malloc 
	--enable-shared-workers 
	--enable-workers 
	--enable-javascript-debugger 
	--enable-fast-mobile-scrolling 
	--enable-offline-web-applications 
	"`
So I would like to compile again web browser to make some attemps, but i don't want build all openatv image!
How can I do?
I know only command
`MACHINE=sf4008 DISTRO=openatv make image`
that build all image

nickersk, 2018-06-28:
cd builds/openatv/release/sf4008
source env.source
to rebuild only one package: MACHINE=sf4008 bitbake -b hbbtv-browser-webkit
to rebuild modified package and image: MACHINE=sf4008 bitbake openatv-image

you can read something like this when you run make without anything else in the top dir

maxbambi, 2018-07-01:
Finally I have compiled webkit browser with svg support...
Unluckily it is not enough to get RAI hbbtv working... 
First of all, a native implementation of bind method is needed...
It has been added in more recent versions of webkit, with a lot of changes compared to that in oe-alliance github, so I gave up.
Using a workaround pasted in javascript code of Rai js script, the home page of hbbtv is shown: 
user can move cursor and select channels or programs in the right way and graphics are updated according to user input.
Sorry execution hangs when you select a program to browse description and watch clips or episodes, because it is shown only a screen with rai logo in a corner and back button....
In console a message appears of an error grabbing info in json format from an url that pasted in browser is working and give the requested output

closed 2018-07-18 by maxbambi