oe-alliance/oe-alliance-core #326: nextp3 - ERROR: enigma2.bb:do_configure

nickersk, 2018-07-07:
getting this:
config.status: error: Something went wrong bootstrapping makefile fragments
     for automatic dependency tracking.  Try re-running configure with the
     '--disable-dependency-tracking' option to at least be able to build
     the package (albeit without support for automatic dependency tracking).

in config.log:
config.status:1930: executing depfiles commands
config.status:2007: cd lib       && sed -e '/# am--include-marker/d' Makefile         | make -f - am--depfiles
/tmp/GmqTB95z:3140: .deps/python/enigma_python.Pcpp: No such file or directory
make: *** No rule to make target '.deps/python/enigma_python.Pcpp'.  Stop

atvcaptain, 2018-07-08:
looks autotools broken, autotools.class use '--disable-dependency-tracking , but looks the option was ignored

nickersk, 2018-07-08:
with --disable-dependecy-tracking fail too .deps/python/enigma_python.Tpo: No such file or directory

atvcaptain, 2018-07-08:
i check first rollback automake to 1.15.1

atvcaptain, 2018-07-08:
test with rollback to 1.15.1 all ok

atvcaptain, 2018-07-08:
i test now revert this commit : http://git.savannah.gnu.org/cgit/automake.git/commit/?id=333b98f2d108111e9a99e7ede17de7f0a1adba03

nickersk, 2018-07-08:
confirmed, no errors with 1.15.1.

nickersk, 2018-07-09:
i suggest rollback for now, because it will nead more time to investigate, but leave this issue open, i'll try to find solution later.

nickersk, 2018-07-09:
same error between 1.16 and 1.16.1

nickersk, 2018-07-10:
issue comes from depout.m4 macro
see https://github.com/oe-alliance/oe-alliance-core/pull/327

closed 2018-07-10 by atvcaptain