oe-alliance/oe-alliance-core #328: python-youtube-dl not work

thawtes, 2018-07-11:
python-youtube-dl not works pls fix it and update latest release
thx

atvcaptain, 2018-07-11:
what not working, how can i test, i never use this

thawtes, 2018-07-11:
hi @atvcaptain 

`root@vuduo2:~# youtube-dl`
`Traceback (most recent call last):`
`File "/usr/bin/youtube-dl", line 6, in <module> from pkg_resources import load_entry_point`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3144, in <module> @_call_aside`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3128, in _call_aside f(*args, **kwargs)`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3157, in _initialize_master_working_set working_set = WorkingSet._build_master()`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 666, in _build_master ws.require(__requires__)`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 984, in require needed = self.resolve(parse_requirements(requirements))`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 870, in resolve raise DistributionNotFound(req, requirers)`
`pkg_resources.DistributionNotFound: The 'youtube-dl==2018.1.21' distribution was not found and is required by the application`

atvcaptain, 2018-07-11:
we update to last version, please check tomorrow agai

thawtes, 2018-07-14:
hello again @atvcaptain 
updated last version but still not work :(

`root@vuduo2:~# youtube-dl`
`Traceback (most recent call last):`
`File "/usr/bin/youtube-dl", line 6, in <module> from pkg_resources import load_entry_point`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3144, in <module>`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3128, in _call_aside`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3157, in _initialize_master_working_set`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 666, in _build_master`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 984, in require`
`File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 870, in resolve`
`pkg_resources.DistributionNotFound: The 'youtube-dl==2018.7.10' distribution was not found and is required by the application`
`root@vuduo2:~#`

atvcaptain, 2018-07-14:
please test

opkg update
opkg install python-distutils

atvcaptain, 2018-07-14:
https://github.com/oe-alliance/oe-alliance-core/commi/028194a39ad2e005b68605018f8d339b8983c82a

 youtube-dl https://youtu.be/KKJVySdBvIY
[youtube] KKJVySdBvIY: Downloading webpage
[youtube] KKJVySdBvIY: Downloading video info webpage
[youtube] KKJVySdBvIY: Downloading MPD manifest
[youtube] KKJVySdBvIY: Downloading MPD manifest
[download] Destination: U5 BT Remote Pairing-KKJVySdBvIY.mp4
[download] 100% of 992.68KiB in 00:00

closed 2018-07-14 by atvcaptain