oe-alliance/oe-alliance-core #371: branch nextp3  oe-alliance-core/meta-oe/recipes-connectivity/samba/pam-smbpass_3.6.25.bb

TwolDE, 2018-08-05:
If I run a clean build (hd51/Giga 4K's) then everything completes fine.
If I rerun the build it fails in compile  samba/pam-smbpass_3.6.25.bb.

So I delete the bb, obviously build fails and then copy the bb back - and everything runs through just fine.

Seems to need to do the samba/pam-smbpass_3.6.25.bb config from afresh.
error in compile are "undefined reference to `_talloc.............."

christophecvr, 2018-08-28:
This issue is same by me. Build from scratch HD51/mutant51 first build samba ok second time fail on compile samba/pam-smbpass_3.6.25.bb samme error "undefined reference to ...

investigating now could be cause on HD51 samba-4.8.4 is used 23 days ago 4.8.3 but same issue.

christophecvr, 2018-08-28:
I don't know exactly, but I do not think so. It's actually only when there was a change into the pam-smbpass. and an update of samba together . 
The pam-smbpass_xxx is oe-alliance the smb used on HD51 is the version into openembedded .
They are different . however if it happens to you and the build is broken due to pam-smbpass.
Just use first command
bitbake samba pam-smbpass -c cleanall  # use all to cleanall the sstate-cache also.
 then rebuild image.
small note: since at this time still works in progress on package manager by openembedded and there are quit a lot of bugs with it, it's not a bad idea to use when you make you're image bitbake -k <you're-image> so at the -k option.

TwolDE, 2018-08-28:
As always many thanks!

nickersk, 2018-09-04:
please see https://github.com/oe-alliance/oe-alliance-core/pull/390

closed 2018-09-04 by atvcaptain