oe-alliance/oe-alliance-core #388: Nextp3 _rootfs build fails

TwolDE, 2018-08-28:
This happens on my Arm receivers since the issue with the 3rd party plugin and no manifest created - this has been fixed but still have the following rootfs issue where it is issuing a remove without any specified package. i-e just remove by itself.

ERROR: openvix-image-6.0-r02S.000 do_rootfs: Unable to remove packages. Command '['/media/OEA/oeaP3/builds/openvix/release/gb7252/tmp/work/gb7252-oe-linux-gnueabi/openvix-image/openvix-image-6.0-r02S.000/recipe-sysroot-native/usr/bin/opkg', '--volatile-cache', '-f', '/media/OEA/oeaP3/builds/openvix/release/gb7252/tmp/work/gb7252-oe-linux-gnueabi/openvix-image/openvix-image-6.0-r02S.000/opkg.conf', '-t', '/media/OEA/oeaP3/builds/openvix/release/gb7252/tmp/work/gb7252-oe-linux-gnueabi/openvix-image/openvix-image-6.0-r02S.000/temp/ipktemp/', '-o', '/media/OEA/oeaP3/builds/openvix/release/gb7252/tmp/rootfs', '--force_postinstall', '--prefer-arch-to-version', '--force-depends', 'remove']' returned 1:
/media/OEA/oeaP3/builds/openvix/release/gb7252/tmp/work/gb7252-oe-linux-gnueabi/openvix-image/openvix-image-6.0-r02S.000/recipe-sysroot-native/usr/bin/opkg: the ``remove'' command requires at least one argument
usage: opkg [options...] sub-command [arguments...]
where sub-command is one of:

Package Manipulation:
	update                          Update list of available packages
	upgrade                         Upgrade installed packages
	install <pkgs>                  Install package(s)
	configure <pkgs>                Configure unpacked package(s)
	remove <pkgs|glob>              Remove package(s)
	clean                           Clean internal cache
	flag <flag> <pkgs>              Flag package(s)
	 <flag>=hold|noprune|user|ok|installed|unpacked (one per invocation)

atvcaptain, 2018-08-28:
read this: https://bugzilla.yoctoproject.org/show_bug.cgi?id=12900

nickersk, 2018-08-28:
... and take a look at this: https://github.com/oe-alliance/build-enviroment/pull/27

christophecvr, 2018-08-28:
@nickersk perhaps nice but not merged yet. But there are also other issues such as the fact that the deploy/ipk/... do not contain a package index any more .  Before this was also created during the rootfs task. Now any more. Means Packages.gz is missing the stb needs it in order to update or obtain the possible plugin list . Ok you can make this self with a recreate-feed script but normally it should have been generated by the package-manager.py during rootfs task.

TwolDE, 2018-08-28:
Thanks guys ...... appreciate all your help -

atvcaptain, 2018-08-28:
yes index file missing i never check it , no idea its new issue or we have it longer

christophecvr, 2018-08-28:
it came at the same time with al the others first the manifest not found due to the ghost search on sdk-native-x86_64-x86_64..... 
The error after the native issue has been worked around by changing bb recipes about : 
the ``remove'' command requires at least one argument

It started as from commit "http://git.openembedded.org/openembedded-core/commit/meta/lib/oe/package_manager.py?id=8ffd93bdb09b0a4a84b27dafcd684c6abba392ed " as nickersk found out

nickersk, 2018-09-17:
please see and test https://github.com/oe-alliance/oe-alliance-core/pull/403

closed 2018-10-07 by atvcaptain