oe-alliance/oe-alliance-core #391: recent oe-core updates

nickersk, 2018-09-06:
so, oe-core switched to openssl 1.1, we have problem with ffmpeg_3 at least, because it will fail in fact librtmp and rtmpdump are uncompilable with openssl 1.1 and therefore are conflicting openssl files installed during do_recipe_sysroot preparation. is anybody able to patch rtmpdump and librtmp to switch to openssl 1.1 ?

atvcaptain, 2018-09-06:
problem with 1.1 was all close binarys use 1.02  when this not abi stable we must rollback to 1,02, or all cams, ci+ broken

arn354, 2018-09-06:
right, forgot about the closed binary blobs relying on openssl 1.02

nickersk, 2018-09-06:
so we should stick with openssl10 for now. right ?

christophecvr, 2018-09-06:
Yes I watched this issue also, Tried to change all oe-metalliance to openssl10  which is then openssl version 1.02. But problem is conflicts since the packages off openembedded are all openssl-1.1 now as a default . Guess both will need to be installed and then like they did with the still openssl-0.79 required for some softcams some compat links patches for libcrypto and libssl.

christophecvr, 2018-09-06:
about librtmp, does someone know for what we need it. There was something about that it was required with the old version of rtmpdump, but not with the new one anymore. 
Seems it is still required.

https://github.com/GStreamer/gst-plugins-bad/tree/master/ext/rtmp

christophecvr, 2018-09-06:
ffmpeg also does need librtmp. As far I know it is required to play iptv rtmp channels.

christophecvr, 2018-09-06:
but source rtmpdump is avbl on oe-alliance. Not sure but could be that librtmp is present in openembedded rtmpdump.

nickersk, 2018-09-06:
rtmpdump in meta-oe is broken with openssl 1.1 maybe is can be possible to split ffmpeg into two packages... make separate bb from bbappend.

nickersk, 2018-09-06:
guys at oe forgot to resolve ssl headers files conflicts. thats why we are facing this...

atvcaptain, 2018-09-07:
at the moment we build again 1.02, for oe-a 4.3 its better we dont switch to 1.1.1, we can open a new branch for test openssl 1.1.1 this need many more time, and we must request new close binary blobs, and mutant und co never send this when pli not use 1.1.1 too

nickersk, 2018-09-07:
i agree, we need no-errors build for ssl10 at first.

closed 2018-09-11 by nickersk