oe-alliance/oe-alliance-core #393: do_unpack fails on cdparanoia recipe

pbasista, 2018-09-09:
Hi, I am trying to compile cdparanoia using [this](../tree/nextp3/meta-oe/recipes-multimedia/cdparanoia/cdparanoia_svn.bb) recipe but it always fails on `unpack` task like this:

```
DEBUG: Executing python function do_unpack
DEBUG: Executing python function base_do_unpack
NOTE: Unpacking /code/openembedded/build-enviroment/sources/cdparanoia_svn.xiph.org_.trunk_17289_.tar.gz to /code/oe-alliance/build-enviroment/builds/openatv/release/sf4008/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/cdparanoia/10.2+svnr17289-r0/

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
ERROR: Unpack failure for URL: 'svn://svn.xiph.org/trunk;module=cdparanoia;protocol=http'. Unpack command PATH="/code/oe-alliance/build-enviroment/openembedded-core/scripts:/code/oe-alliance/build-enviroment/builds/openatv/release/sf4008/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/cdparanoia/10.2+svnr17289-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi:/code/oe-alliance/build-enviroment/builds/openatv/release/sf4008/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/cdparanoia/10.2+svnr17289-r0/recipe-sysroot/usr/bin/crossscripts:/code/oe-alliance/build-enviroment/builds/openatv/release/sf4008/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/cdparanoia/10.2+svnr17289-r0/recipe-sysroot-native/usr/sbin:/code/oe-alliance/build-enviroment/builds/openatv/release/sf4008/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/cdparanoia/10.2+svnr17289-r0/recipe-sysroot-native/usr/bin:/code/oe-alliance/build-enviroment/builds/openatv/release/sf4008/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/cdparanoia/10.2+svnr17289-r0/recipe-sysroot-native/sbin:/code/oe-alliance/build-enviroment/builds/openatv/release/sf4008/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/cdparanoia/10.2+svnr17289-r0/recipe-sysroot-native/bin:/code/oe-alliance/build-enviroment/bitbake/bin:/code/oe-alliance/build-enviroment/builds/openatv/release/sf4008/tmp/hosttools" tar xz --no-same-owner -f /code/openembedded/build-enviroment/sources/cdparanoia_svn.xiph.org_.trunk_17289_.tar.gz failed with return value 2
DEBUG: Python function base_do_unpack finished
DEBUG: Python function do_unpack finished
ERROR: Function failed: base_do_unpack
```

The issue is caused by the fact that the file containing the source code (in my case `/code/openembedded/build-enviroment/sources/cdparanoia_svn.xiph.org_.trunk_17289_.tar.gz`) is *not* a gzipped tar archive, but rather an *uncompressed* tar archive, despite having the `.gz` file extension. Bitbake tries to extract it using `tar xz` command, which causes the error.

I believe that cdparanoia recipe needs to be updated to take that into account, but I do not know how to do it. The recipe itself does not seem to contain any configuration related to file name or extraction method. Moreover, it is supposed to get the sources from svn, so I am not even sure how the `tar.gz` file was created in the first place.

BTW, simply changing the revision in the recipe file to the newest available svn revision `19649` helps. Bitbake now fetches the sources archive fine, because the fetched archive for revision `19649` is actually a gzip-compressed tar archive.

But I believe perhaps bitbake fetch task should be more robust against files with incorrect suffixes and it should be able to extract them properly. A trivial solution would be for bitbake to let `tar` determine the need to use compression and not to try to infer the file type and supply the `z` or similar command line options to `tar`.

nickersk, 2018-09-09:
have you tried rebuild with bitbake -c cleanall -b cdparanoia_svn and bitbake -b cdparanoia_svn ?

atvcaptain, 2018-09-09:
or delete only the code/openembedded/build-enviroment/sources/cdparanoia_svn.xiph.org_.trunk_17289_.tar.gz in your source folder and bb fetch new, logs file was currupt

nickersk, 2018-09-09:
however, updated to latest svn, check https://github.com/oe-alliance/oe-alliance-core/pull/394

pbasista, 2018-09-09:
Thank you. I must have left some old files present in the build directory. I have been playing with this package before, but I have since forgotten about it.

After `bitbake -c cleanall`, the package is fetched correctly, even on older revision. Sorry for raising an issue before double checking the setup on my side.

And also, thanks for updating the package to its newest revision.

closed 2018-09-09 by atvcaptain