oe-alliance/oe-alliance-core #411: Audio Problems

DennyCoy, 2018-10-27:
Hello, all...

I have a problem in the Dinobot 4K box on the SD channel on the Videocon Provider ( Satelite ST2 ku-band ) with the cccam server ...
The problem is that SD videocon channels do not come out Audio when HD Audio channels are not a problem ... as a consideration I tried using the  Formuler F4 Turbo box with openATV6.2 SD channel audio and HD audio channels have no problems .... I have set oscam Dinobot 4K and equates the parameters with the Formuler F4 Turbo but it doesn't work
I use the OpenATV 6.2 image on the Dinobot 4K box ...
please give a solution ... thank you

atvcaptain, 2018-10-28:
send record file 100 mb with https://wetransfer.com/

DennyCoy, 2018-10-28:
Please sir ... the file record link
https://we.tl/t-Dwisu2RLSh

atvcaptain, 2018-10-28:
ok i send the infos to dev team

DennyCoy, 2018-10-28:
ok sir ... thank you very much for the time

atvcaptain, 2018-10-28:
play with vlc player no audio, have you other e2 box can can make a record and check audio works and send this file too

DennyCoy, 2018-10-28:
Ok Sir....
This is an example of a file record from the Formuler F4 Turbo box .... Audio Channel SD doesn't matter...
https://we.tl/t-bXbftlYfwM

atvcaptain, 2018-10-28:
ok thx audio with vlc works, looks issue decrypt audio with dino

closed 2019-01-09 by arn354