oe-alliance/oe-alliance-core #420: ZGEMMA H9S CANNOT OPEN ENCRYPTION OF POWERVU DES

DennyCoy, 2018-11-15:
Dear All....

Please help with the zgemma H9S Enigma2 box, which cannot open powervu DES encryption. Maybe it needs an update DRIVER. I use the openATV 6.2 image
for reference I include attachments
thank you
[image]
[image]

GigaBlue-STB, 2018-11-15:
See #412

closed 2018-11-15 by arn354

DennyCoy, 2018-11-16:
Not Work Sir....
I tried a few oscam parameters but the channel is still dark...
for channels with powervu CSA encryption it doesn't have a problem ... but some channels with powervu DES encryption don't work...
I used to have the same problem with the Formuler F4 Turbo box, it can't open PowerVu DES ... the problem is solved by updating the driver ...
could a case like this be the same...
[image]
[image]
[image]