oe-alliance/oe-alliance-core #421: Zgemma H9S Cannot Open Encryption Of Powervu DES

DennyCoy, 2018-11-17:
Referring to issue #420 powervu DES encryption still does not work... 
I tried a few oscam parameters but the channel is still dark...
for channels with powervu CSA encryption it doesn't have a problem ... but some channels with powervu DES encryption don't work...
I used to have the same problem with the Formuler F4 Turbo box, it can't open PowerVu DES ... the problem is solved by updating the driver ...
could a case like this be the same...
thank you..

DennyCoy, 2018-11-21:
Hello...

arn354, 2018-11-21:
We can't update a driver if a manufacturer does not deliver it. Please ask the manufacturer.

DennyCoy, 2018-11-21:
Ok sir ... thank you ... I tried asking the manufacturer

nautilus7, 2019-03-26:
@arn354 wouldn't be easier for you as image developers (than each independent user) to put pressure to the manufacturer and ask for new drivers??? The manufacturer provides you with the divers after all...

arn354, 2019-03-26:
I only have contact to GigaBlue.

atvcaptain, 2019-03-26:
we have allready send this info to driver team, driver team say all ok

DennyCoy, 2019-03-29:
thanks for the attention sir ... for powervu des while i use stream relay ...