oe-alliance/oe-alliance-core #442: e2 compile error after glibc initial removed from oecore

nickersk, 2018-12-26:
after glibc initial was removed from oecore i get this error while do_compile  enigma2.bb 
:  crt1.o: undefined reference to symbol '_gp'
i am playing with it few days, but still have no idea what has to be fixed ...

atvcaptain, 2018-12-26:
found this, https://patchwork.ozlabs.org/patch/312275/  

but no idea its same issue with openssl

i start now build fresh need some time before e2 build start

nickersk, 2018-12-26:
building oatv image for vusolose, looks like libv3ddriver.so is causing this issue (it is part of graphics drivers). other mipsel machines can be affected too ... probably.

atvcaptain, 2018-12-27:
i build osmio4k use too libv3 addon from edision but no build issue, i start build now vusolose too

nickersk, 2018-12-27:
pls update oecore to latest master first - glibc initial was removed after commit 956 ...

nickersk, 2018-12-27:
it's related to this maybe: https://bugzilla.yoctoproject.org/show_bug.cgi?id=13097

atvcaptain, 2018-12-27:
now i have many fetch errors

  /home/oe1/nextp3/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-channelsettings/enigma2-plugin-settings-henksat-13e-19e-23e-28e-42e-7e.bb:do_fetch
  /home/oe1/nextp3/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-channelsettings/enigma2-plugin-settings-gigablue-wilhelmtel.bb:do_fetch
  /home/oe1/nextp3/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/e2openplugins/enigma2-plugin-extensions-permanentvfdclock.bb:do_fetch
  /home/oe1/nextp3/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-channelsettings/enigma2-plugin-settings-vhannibal.motor.and.dtt.roma.bb:do_fetch
  /home/oe1/nextp3/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-channelsettings/enigma2-plugin-settings-vhannibal.dual.feeds.and.dtt.torino.bb:do_fetch

nickersk, 2018-12-27:
this is related to sqlite db setup in bitbake (oe guys changed that few dayd ago). u need to run bitbake more times to get through this locking protocol errors...

atvcaptain, 2018-12-27:
build arm fresh with last master and finish without error

atvcaptain, 2018-12-28:
https://github.com/oe-alliance/oe-alliance-core/commit/31d881bcd6de75e87de6c0baed03e826f0362f02

build running , i have first remove uianimation only vu have it

nickersk, 2018-12-28:
vusolose e2 compiled ok without uianimation

atvcaptain, 2018-12-28:
for the fetch issue i add a bug: https://bugzilla.yoctoproject.org/show_bug.cgi?id=13108

nickersk, 2019-01-08:
fetch patch is in master branch now, so we can update OE submodules ....

closed 2019-01-08 by atvcaptain