oe-alliance/oe-alliance-core #458: wget issues Nextp3 branch

davesayers2014, 2019-01-23:
Hi I have been having certificate issues with OpenATV 6.3 as reported here https://www.opena.tv/english-section/45237-wget-changes-openatv-6-3-a.html

To get around this I have used --no-check-certificate to download files.

Sadly this doesn't get around opkg install from websites ie  opkg install --force-reinstall "https://github.com/su1s/e2m3u2bouquet/releases/download/v0.8.2/enigma2-plugin-extensions-e2m3u2bouquet_0.8.2_all.ipk"

I tried adding --no-check-certificate to opkg install but it complains that command doesnt exist.

davesayers2014, 2019-01-24:
Ok only way ive got round this is via downloading a .wgetrc within this I have check_certificate = off https://raw.githubusercontent.com/davesayers2014/e2m3ubouquet/master/files/.wgetrc

So I can use --no-check-certificate to download .wgetrc then I can install ipk,s without issues.

`wget -O /home/root/.wgetrc --no-check-certificate "https://raw.githubusercontent.com/davesayers2014/e2m3ubouquet/master/files/.wgetrc" && chmod +x .wgetrc && opkg install --force-reinstall "https://github.com/su1s/e2m3u2bouquet/releases/download/v0.8.2/enigma2-plugin-extensions-e2m3u2bouquet_0.8.2_all.ipk`

Or just use curl

It would be better if these wget changes was sorted though.

davesayers2014, 2019-02-16:
I'll close this issue as it's not been looked at, I have the above solution other users will need to work it out. Thanks for the help

davesayers2014, 2019-02-16:
New log reported by s3n0

https://www.opena.tv/english-section/45237-wget-changes-openatv-6-3-a-post388987.html#post388987

nickersk, 2019-02-16:
i will take a look @this

nickersk, 2019-02-17:
i think we are dealing with update-ca-certificates script problem, because since openssl 1.1 it is using 'openssl rehash' command, which is not available in openssl 1.0. so it fails to rehash certs during do_rootfs ....

Schimmelreiter, 2019-02-17:
Sounds plausible to me.

nickersk, 2019-02-17:
please see and test https://github.com/oe-alliance/oe-alliance-core/commit/c2ed92a72dd1097b655309e6d4061de21e7015b3

davesayers2014, 2019-02-17:
Thanks I'll test on next nightly OpenATV 6.3 release or is there a way to add this to current image?

nickersk, 2019-02-17:
you can try to build and reinstall openssl and ca-certificates pkgs

nickersk, 2019-02-17:
6.3 beta images are built from 4.3 branch afaik, so this solution have to be tested in nextp3 and backported to  4.3 branch later.

davesayers2014, 2019-02-19:
Sorry I dont know how to buiild image and I thought OpenATV 6.3 was build fro oe core Nextp3 branch. It must have changed from Nextp3 to 4.3 at some point.

If someone wants to test this they can try by opkg install any plugin from third party feeds github and wget tmp any file on github.

nickersk, 2019-02-25:
no more error reports, so closing this issue.

closed 2019-02-25 by nickersk

davesayers2014, 2019-02-25:
So has this been implemented to oe 4.3 branch? https://github.com/oe-alliance/oe-alliance-core/pull/472

I opened this issue report thinking OpenATV was built on Nextp3 branch as I'm sure it originally was but it might have changed at some point. Thanks.

nickersk, 2019-02-25:
pls see https://github.com/oe-alliance/oe-alliance-core/commit/7806bd74cf8932dadc3b1bf549d43421e62f096f

davesayers2014, 2019-02-25:
Thanks I've just tested by flashing latest OpenATV 6.3 using openmultiboot and used opkg install and wget from github and had no certificate issues. Thank you for the fix.