oe-alliance/oe-alliance-core #471: cube build cross-localedef

atvcaptain, 2019-02-19:
have any a idea before i start google search

Exception: subprocess.CalledProcessError: Command '['cross-localedef', '--verbose', '--uint32-align=4', '--little-endian', '--add-to-archive', '/www/www/atv63arm/build-enviroment/builds/openatv/release/cube/tmp/rootfs/cube/usr/lib/locale/uk_UA']' returned non-zero exit status 1

Subprocess output:
cannot create temporary file: /www/www/atv63arm/build-enviroment/builds/openatv/release/cube/tmp/work/x86_64-linux/cross-localedef-native/2.23-r0/recipe-sysroot-native/usr/lib/locale/locale-archive.9IzCmF: No such file or directory

ERROR: openatv-image-6.3-0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /www/www/atv63arm/build-enviroment/builds/openatv/release/cube/tmp/work/cube-oe-linux-gnueabi/openatv-image/6.3-0/temp/log.do_rootfs.52841
NOTE: recipe openatv-image-6.3-0: task do_rootfs: Failed
ERROR: Task (/www/www/atv63arm/build-enviroment/meta-oe-alliance/meta-oe/recipes-distros/openatv/image/openatv-image.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 13264 tasks of which 12782 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /www/www/atv63arm/build-enviroment/meta-oe-alliance/meta-oe/recipes-distros/openatv/image/openatv-image.bb:do_rootfs

nickersk, 2019-02-25:
same result here:
Exception: subprocess.CalledProcessError: Command '['cross-localedef', '--verbose', '--uint32-align=4', '--little-endian', '--add-to-archive', '/oea/4.3/builds/openatv/release/cube/tmp/rootfs/cube/usr/lib/locale/sl_SI']' returned non-zero exit status 1.

Subprocess output:
cannot create temporary file: /oea/4.3/builds/openatv/release/cube/tmp/work/x86_64-linux/cross-localedef-native/2.23-r0/recipe-sysroot-native/usr/lib/locale/locale-archive.bcnqXE: No such file or directory

going to do research to find a solution for us.

nickersk, 2019-02-25:
should be fixed now:

pls see https://github.com/oe-alliance/oe-alliance-core/commit/47cba5d3bedc33d5289606738d7b6af9d4eb3852

nickersk, 2019-02-27:
can we close this as resolved ?

atvcaptain, 2019-02-27:
sure fixed thx for help

closed 2019-02-27 by atvcaptain