oe-alliance/oe-alliance-core #473: Blindscan problem with USALS

ei4kn, 2019-02-21:
When I try using blindscan the dish moves to an incorrect position (aprroximately 16 degrees west of where it should be).  It works correctly using Openvix build 5.3.024 which uses the 10th January blindscan plugin.  The plugin was changed on 16th January and this broke blindscan.  Manual scan works perfectly so I believe the problem is in the octagon-blindscan module somewhere.

Huevos, 2019-02-21:
The plugin does not control dish position. That is done by the image and drivers.

ei4kn, 2019-02-21:
The co-ordinates are correct.  With Vix 024 it works fine and I cant find any other build/image that works. Openatv dont seem to have an archive going back to January to test an image.  Manual scan works correctly as does automatic scan and there are no issues watching tv.  I've never had issues with my solo2 or my uno before that.

ei4kn, 2019-02-21:
But it may be something to do with longtitude all the same.  I changed my position to 8 East instead of 8 West - I lost picture of course.  Then I did a blindscan on 28E and again picked up channels on Hotbird. So changing from West to East made no difference on blindscan.  For blindscan the longtitude in being treated as east regardless.  Maybe this is a clue of where the problem lies.  Thanks for your help (and the blindscan plugin development!)

Huevos, 2019-02-22:
8 west? Where do you live? Ireland?

ei4kn, 2019-02-22:
Correct - I'm near the centre of Ireland.  And for people in Germany etc the problem will not occur because east positions work correctly..

Huevos, 2019-02-22:
I'm baffled because I am west of zero and my USALS works fine. Using OpenViX 029.

ei4kn, 2019-02-22:
USALS is working fine for me too for TV viewing, and for manual/automatic scan.  Its only blindscan has the problem.

ei4kn, 2019-02-23:
Just tried with the 220219 drivers and no change.   I tried something different to see what is happening.  I selected a channel on 23.5e before going to blindscan.  In the blindscan menu I then selected 28.2e and the dish moved east to where it should go and "lock" came up on screen - all good so far.  But once I hit the green button to scan for transponders the dish moved west.  So it is when it is actually scanning not before the scan that the wrong commands are sent to the motor.

ei4kn, 2019-02-23:
@heuvos - you may be west but if you are only maybe 1 degree west and the dish is not too big it may actually still work for you.

Huevos, 2019-02-23:
So now you are telling me my install is not working properly and I haven't noticed? LOL.

ei4kn, 2019-02-23:
I didnt mean to insult you.  Ive been using e2 receivers for years without problems like this.  I think maybe I should just send it back to WOS as it looks like this issue is not going to be fixed.  As I have stated many times it works fine with Vix 024 so my hardware should be fine.  I plugged back in my Solo2 which luckily I had kept and blindscan works fine (software blinscan).

Huevos, 2019-02-23:
What receiver is it

AbuBaniaz, 2019-02-23:
Octagon Sf8008

This is Ronand on our forum, I guess

ei4kn, 2019-02-23:
Sorry its an Octagon SF8008. Yes thats me

Huevos, 2019-02-23:
https://github.com/oe-alliance/oe-alliance-core/commit/be06653bf0f2d9e6e3b06788444095d3ca8d8f0b

That is latest version. For me it works fine for blindscan. But not in 029. Will be in 030.

ei4kn, 2019-02-23:
Hopefully. I did a dev build of 030 a few minutes ago but no change but will wait till released.

Huevos, 2019-02-23:
Ok so take the octagon-blindscan from 024 and drop it in your current build.

ei4kn, 2019-02-23:
That doesnt seem to work - It doesnt find any transponders but doesnt actually seem to do any scanning as it gives the message pretty instantyly

Huevos, 2019-02-23:
Ok. Can't be compatible then. I'll have to think of a test to prove or disprove.

ei4kn, 2019-02-23:
No problem.  Its saturday night - drink some beer instead!

Huevos, 2019-02-24:
I confirm this bug. 

Set location to 008.000 WEST in tuner config. Open blindscan plugin. Select satellite 53.0 east. Motor moves to correct location and stops. Press OK to start the scan. Dish moves several degrees to the west. Scan fails because dish is pointing in the wrong direction.

I've reported to Octagon.

ei4kn, 2019-02-24:
Many thanks Heuvos. At least I know now that I'm not going mad haha.  Well maybe I am but at least this is not the cause.

ei4kn, 2019-02-24:
Just as a side note - is it necessary to send the motor commands while blinscanning even though the dish would already have moved to the correct position?  Or is that just the way it works?

Huevos, 2019-02-24:
It is just a couple of receivers that do this. Most receivers let the plugin (enigma) control the tuning and the executable is passive. In the case of the sf8008 the frontend is closed and the executable re-opens it. That means all the DiSEqC code in enigma has to be duplicated in the blindscan executable.

https://github.com/oe-alliance/oe-alliance-plugins/blob/master/Blindscan/src/plugin.py#L915-L916

ei4kn, 2019-02-24:
OK thanks for that. Hopefully Octagon will sort it.  I know most people wont care for blindscan but I like feedhunting and its one of the reasons I picked the SF8008 over the zgemma h7s as it doesnt have blinscan AFAIK (as well as more flash memory).

Huevos, 2019-02-24:
H9 has blindscan so I'm sure H7 has too. Got one here but not connected at the moment to check. But it does not use the same plugin, so you lose all the filters of the blindscan plugin. And maybe all the other stuff like C-band and Ka band scanning is also not available.

ViX-Sicilian, 2019-02-25:
H7 doesn't currently support blindscan. H9 does.

esmos, 2019-02-25:
blindscna usals please use today octagon-blindscan file, replace by ftp in usr/bin folder.

ei4kn, 2019-02-25:
I have tried the new file with Openvix 028,029 and dev build on 030 and there is no change.  I also tried todays OpenATV 6.2.  Maybe Heuvos can confirm the problem remains.

AbuBaniaz, 2019-02-26:
As this is a manufacturer issue, and they have supplied a fixed binary, can we close this issue?

ei4kn, 2019-02-26:
Yes you can close it. Thanks.

closed 2019-02-27 by atvcaptain