oe-alliance/oe-alliance-core #476: Support BTRFS

petrkr, 2019-02-24:
Hello,
can you consider add BTRFS filesystem support, can be as optional kmod module for later install.

I have one external hard-drive with that FS and also it could be good for mainFS (same like turris have) for revisions and snapshots.

Regards Petr

nickersk, 2019-02-25:
i think this can only be done per individual machine defconfig. or am i wrong?

petrkr, 2019-02-25:
I will check that, can be. Whole build system style is kinda new for me. Most of images are somehow hidden or unclear who do what. I now use PurE2 images and their git seems to be force-pushed. Someone who know will know in few minutes where to put kernel config =M

Thanks

nickersk, 2019-02-25:
enabling this feature in kernels prior to  4.x series should be double thinked as many sites are reporting strange or unwanted BTRFS behaviour enabled in older kernels ...

petrkr, 2019-02-25:
I do not say switch on it, just supports it, octagon already runs on 4.1.x kernel. Turris omnia runs on btrfs whole FS and I run on laptop for it more than year too. But as I said, for now it would be enough make it as module and install-able same like dvb or wifi modules.

nickersk, 2019-02-25:
well, building kernel module to be alone installable package outside kernel source tree requires source files. if you enable it as M in defconfig nothing needs to be done more, u can load it on running system by modload .... maybe some1 got more knowledge so it is needed to be discussed more for sure.

petrkr, 2019-10-19:
Hello, how looks support for BTRFS FS ? I now thinking also it could be rootfs in some experimental images for better multi-image and multi-boot cases. since there can be subvolumes as different images. I do not have binary blobs for my image, so I can not recompile kernel image myself :(

But probably for boot support I will need =Y config than =M if there is no initramfs

petrkr, 2021-08-12:
Switched to openvision makes this request for me pointless, closing issue.

closed 2021-08-12 by petrkr