oe-alliance/oe-alliance-core #486: nextp3 buid broken

atvcaptain, 2019-04-06:
e_prepare_recipe_sysroot.tgz.siginfo', runtime='customfile:/www/www/nextp3/build-enviroment/builds/openatv/release/hd51/tmp/stamps/cortexa15hf-neon-vfpv                                      4-oe-linux-gnueabi/enigma2-plugin-skincomponents-mediaportal-smoke-hd-skin/1.0+gitAUTOINC+ad888f6981-r0'):
                 os.chmod(tmpfile, 0o664)
    >            os.rename(tmpfile, sigfile)
             except (OSError, IOError) as err:
OSError: [Errno 36] File name too long: '/www/www/nextp3/build-enviroment/builds/openatv/sstate-cache/e7/sigtask.7dmlm8v3' -> '/www/www/nextp3/build-env                                      iroment/builds/openatv/sstate-cache/e7/sstate:enigma2-plugin-skincomponents-mediaportal-smoke-hd-skin:cortexa15hf-neon-vfpv4-oe-linux-gnueabi:1.0+gitAUT                                      OINC+ad888f6981:r0:cortexa15hf-neon-vfpv4:3:e7083f7182630af30cef2634260c2a267cb6e8d054dff421b1d187bd94a434fe_prepare_recipe_sysroot.tgz.siginfo'

nickersk, 2019-04-06:
those recipes names like enigma2-plugin-skincomponents-mediaportal-smoke-hd-skin in this case must be shortened to something shorter or version in recipe must not be set to autorev.

atvcaptain, 2019-04-06:
ext4 max 255  here now 321 not good

atvcaptain, 2019-04-06:
https://bugzilla.yoctoproject.org/show_bug.cgi?id=13268

atvcaptain, 2019-04-06:
netxp3 any news about issue mipsel , you know we must disaable vu animation

nickersk, 2019-04-06:
havent try it for a while. will try... but no news till now

nickersk, 2019-04-06:
imho it is a problem of libv3ddriver.so from libgles package. so it is "fixable" by vuplus guys only.

nickersk, 2019-04-06:
i've found that if i remove GLES_CFLAGS="-DUSE_LIBVUGLES2" from configure.ac build goes ok.

nickersk, 2019-04-06:
e2 wont start: 
Traceback (most recent call last):
  File "/usr/lib/enigma2/python/mytest.py", line 8, in <module>
    from Tools.Profile import profile, profile_final
  File "/usr/lib/enigma2/python/Tools/Profile.py", line 3, in <module>
  File "/usr/lib/enigma2/python/Tools/Directories.py", line 4, in <module>
  File "/usr/lib/enigma2/python/enigma.py", line 3584, in <module>
AttributeError: 'module' object has no attribute 'eDVBCIInterfaces_announceCCK'

nickersk, 2019-04-07:
i am unable to find from where this comes, it is mentioned in enigma2.py only:
eDVBCIInterfaces.announceCCK = new_instancemethod(_enigma.eDVBCIInterfaces_announceCCK, None, eDVBCIInterfaces)

nickersk, 2019-04-15:
https://bugzilla.yoctoproject.org/show_bug.cgi?id=13268#c1  we have to decide how to modify bb's to close this issue until next OE milestone 1 release. should we shorten recipe names or versions inside recipes?

atvcaptain, 2019-04-15:
we must check plugin browser , i hope we dont must change all e2 names

atvcaptain, 2019-05-28:
a solution for vu ani ? https://github.com/openembedded/openembedded-core/commit/a6f9327e79df1bfadb1f613eda799586e60009f2

nickersk, 2019-05-28:
will try ...

nickersk, 2019-05-31:
https://github.com/oe-alliance/oe-alliance-core/commit/89502824584545156ccde560eb6ce00ba30507e1

nickersk, 2019-06-04:
any news about long file names ?

atvcaptain, 2019-06-05:
no news but this month i replace all my build servers next branch build must now wait, next month i can test more

nickersk, 2019-06-05:
ok, no problem, just let me know then ... closing for now ..

closed 2019-06-05 by nickersk