oe-alliance/oe-alliance-core #487: Zgemma H7s CPU temperature

Marchella1983, 2019-04-07:
hi, please if possible add the ability to output the CPU temperature
`root@h7:~# cat /sys/devices/virtual/thermal/thermal_zone0/temp
cat: read error: Invalid argument
root@h7:~#`

on ET 11000 this works with Broadcom BCM7251S
`root@et1x000:~# cat /sys/devices/virtual/thermal/thermal_zone0/temp
55017
root@et1x000:~#`

atvcaptain, 2019-04-07:
driver team have remove on request from sales shops , user read box to hot  and send box back

Marchella1983, 2019-04-07:
Captain I don't understand You very well... Does that mean it won't happen?

koivo, 2019-04-07:
This was a request from shops. They want to have this feature.
Too many users has send back the boxes to shops.

Marchella1983, 2019-04-08:
Yes but why remove when the Broadcom chip BCM7251S is able to do, I also heated the chip but I made myself so
[image]

and I want to have a temperature output room `cat /sys/devices/virtual/thermal/thermal_zone0/temp`

jenseneverest, 2019-04-08:
am i correct that it would be possible if i were to edit the defconfig ?? For a home build.....

`CONFIG_THERMAL=y
CONFIG_THERMAL_OF=y
# CONFIG_THERMAL_WRITABLE_TRIPS is not set
CONFIG_THERMAL_DEFAULT_GOV_STEP_WISE=y
# CONFIG_THERMAL_DEFAULT_GOV_FAIR_SHARE is not set
# CONFIG_THERMAL_DEFAULT_GOV_USER_SPACE is not set
# CONFIG_THERMAL_DEFAULT_GOV_POWER_ALLOCATOR is not set
# CONFIG_THERMAL_GOV_FAIR_SHARE is not set
CONFIG_THERMAL_GOV_STEP_WISE=y
# CONFIG_THERMAL_GOV_BANG_BANG is not set
# CONFIG_THERMAL_GOV_USER_SPACE is not set
# CONFIG_THERMAL_GOV_POWER_ALLOCATOR is not set
CONFIG_CPU_THERMAL=y
# CONFIG_CLOCK_THERMAL is not set
# CONFIG_THERMAL_EMULATION is not set
# CONFIG_QORIQ_THERMAL is not set`

atvcaptain, 2019-04-08:
bcm driver must enable this, wothout we get no infos from the sensor, and driver team have removed this on request from sales

Marchella1983, 2019-04-08:
**atvcaptain** can then please give the contact details of the driver team?

atvcaptain, 2019-04-08:
send mail to zgemma

Marchella1983, 2019-04-08:
thanks, I tried to do that, but they're not answering...

opfans, 2019-04-10:
i have the  H7S box .it works very OK in the ari-condition room.

Marchella1983, 2019-04-10:
I just ask you to add output with the command

I have now at room temperature 24 degrees Zgemma H7s is gaining almost 60 degrees ...

and what will happen when we have summer heat in Uzbekistan reaches 60 degrees of heat ...

write again just add a conclusion
`cat /sys/devices/virtual/thermal/thermal_zone0/temp`
what can Broadcom chip BCM7251S

opfans, 2019-04-16:
> I just ask you to add output with the command
>
> I have now at room temperature 24 degrees Zgemma H7s is gaining almost 60 degrees ...
>
> and what will happen when we have summer heat in Uzbekistan [...]

does people can live with 60 degrees?do not worry ,i have tested h7s worked OK in the room with temperature 60 degrees for 2 months.if you care the high temperature ,install air conditioning equipment.i feel very hot with 38 degrees in china.

Marchella1983, 2019-04-16:
> Can people live with 60 degrees?do not worry ,i have tested h7s worked OK in the room with temperature 60 degrees for 2 months.if you care the high temperature ,install air conditioning equipment.i feel very hot with 38 degrees in china.

You misunderstand me!!!  I only ask to add that Broadcom chip BCM7251S CAN, and it is a conclusion cat /sys/devices/virtual/thermal/thermal_zone0/temp...  here on Zgemma H9 Combo it displays...
[image]

closed 2021-12-24 by atvcaptain