oe-alliance/oe-alliance-core #491: add-gstplayer-support not work !

thawtes, 2019-04-17:
hello,
kodi add-gstplayer-support.patch work openATV 6.2 but not work openATV 6.3
 pls fix thanks...

thawtes, 2019-04-17:
kodi log message

`/usr/bin/kodi: line 195:  4535 Segmentation fault      "$LIBDIR/${bin_name}/${bin_name}.bin" $SAVED_ARGS`

atvcaptain, 2019-04-18:
ok we remove the patch thx for the info

thawtes, 2019-04-18:
> 
> 
> ok we remove the patch thx for the info

this is a very nice feature and works very nicely with 4.2 branch build.
can't you make it work with 4.3 brach build instead of removing this patch?

68foxboris, 2019-07-03:
Not work gstplayer on kodi,what can be done?
root@et1x000:~# init 4
root@et1x000:~# kodi
Warning: MHD_USE_THREAD_PER_CONNECTION must be used only with MHD_USE_INTERNAL_POLLING_THREAD. Flag MHD_USE_INTERNAL_POLLING_THREAD was added. Consider setting MHD_USE_INTERNAL_POLLING_THREAD explicitly.
Warning: MHD_USE_THREAD_PER_CONNECTION must be used only with MHD_USE_INTERNAL_POLLING_THREAD. Flag MHD_USE_INTERNAL_POLLING_THREAD was added. Consider setting MHD_USE_INTERNAL_POLLING_THREAD explicitly.
ERROR: Exception caught on main loop. Exiting
CGstPlayerVideo::handleMessage: Message STATE_CHANGED
CGstPlayerVideo::handleMessage: Element streamsynchronizer0 changed state from NULL to READY.
CGstPlayerVideo::handleMessage: Message STATE_CHANGED
CGstPlayerVideo::handleMessage: Element playsink changed state from NULL to READY.
CGstPlayerVideo::handleMessage: Message STATE_CHANGED
CGstPlayerVideo::handleMessage: Element playbin changed state from NULL to READY.
CGstPlayerVideo::handleMessage: Message STATE_CHANGED
CGstPlayerVideo::handleMessage: Element streamsynchronizer0 changed state from READY to PAUSED.
CGstPlayerVideo::handleMessage: Message STATE_CHANGED
CGstPlayerVideo::handleMessage: Element uridecodebin0 changed state from NULL to READY.
CGstPlayerVideo::handleMessage: Message STATE_CHANGED
CGstPlayerVideo::handleMessage: Element typefindelement0 changed state from NULL to READY.
CGstPlayerVideo::handleMessage: Message STATE_CHANGED
CGstPlayerVideo::handleMessage: Element typefindelement0 changed state from READY to PAUSED.
CGstPlayerVideo::handleMessage: Message STATE_CHANGED
CGstPlayerVideo::handleMessage: Element source changed state from NULL to READY.
CGstPlayerVideo::handleMessage: 0x92579b88A
/usr/bin/kodi: line 195: 29554 Segmentation fault      "$LIBDIR/${bin_name}/${bin_name}.bin" $SAVED_ARGS
[kodi.zip](https://github.com/oe-alliance/oe-alliance-core/files/3356143/kodi.zip)

thawtes, 2019-07-03:
> Not work gstplayer on kodi,what can be done?
> root@et1x000:~# init 4 root@et1x000:~# kodi
> Warning: MHD_USE_THREAD_PER_CONNECTION must be used only with MHD_USE_INTERNAL_POLLING_THREAD. Flag [...]

work on openATV 6.2

closed 2019-07-03 by thawtes

68foxboris, 2019-07-04:
This and I know but how to make it work on 6.3?