oe-alliance/oe-alliance-core #495: BOD - HbbTV crash since a few days

jbleyel, 2019-05-01:
4:00:32.7272 [   ] dvb/pmt.cpp:466 AITready [eDVBServicePMTHandler] Found : control[1], name[ZDF Start], url[http://hbbtv.zdf.de/zdfstart/index.php]
04:00:32.7272 [   ] dvb/pmt.cpp:466 AITready [eDVBServicePMTHandler] Found : control[2], name[ZDF mediathek], url[http://hbbtv.zdf.de/zdfm3/index.php]
04:00:32.7273 [   ] dvb/pmt.cpp:466 AITready [eDVBServicePMTHandler] Found : control[2], name[ZDF Nachrichten], url[http://hbbtv.zdf.de/zdfnews/index.php]
04:00:32.7273 [   ] dvb/pmt.cpp:466 AITready [eDVBServicePMTHandler] Found : control[2], name[ZDF Programm], url[http://hbbtv.zdf.de/zdfepg/index.php]
04:00:32.7273 [   ] dvb/pmt.cpp:466 AITready [eDVBServicePMTHandler] Found : control[2], name[ZDF Fernsehgarten 360°], url[http://d6y27g59ml3bp.cloudfront.net/fernsehgarten/live/index.html]
04:00:32.7308 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Name  : ZDF Start
04:00:32.7313 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] URL   : http://hbbtv.zdf.de/zdfstart/index.php
04:00:32.7318 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] OrgID : 17
04:00:32.7323 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] AppID : 1
04:00:32.7328 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Control Code : 1
04:00:32.7333 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Profile Code : 0
04:00:32.7337 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] 
04:00:32.7342 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Name  : ZDF mediathek
04:00:32.7346 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] URL   : http://hbbtv.zdf.de/zdfm3/index.php
04:00:32.7351 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] OrgID : 17
04:00:32.7356 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] AppID : 2
04:00:32.7360 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Control Code : 2
04:00:32.7368 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Profile Code : 0
04:00:32.7372 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] 
04:00:32.7377 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Name  : ZDF Nachrichten
04:00:32.7382 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] URL   : http://hbbtv.zdf.de/zdfnews/index.php
04:00:32.7386 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] OrgID : 17
04:00:32.7391 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] AppID : 7
04:00:32.7396 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Control Code : 2
04:00:32.7400 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Profile Code : 0
04:00:32.7405 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] 
04:00:32.7409 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Name  : ZDF Programm
04:00:32.7414 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] URL   : http://hbbtv.zdf.de/zdfepg/index.php
04:00:32.7419 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] OrgID : 17
04:00:32.7424 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] AppID : 6
04:00:32.7428 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Control Code : 2
04:00:32.7433 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Profile Code : 0
04:00:32.7437 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] 
04:00:32.7443 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] Traceback (most recent call last):
04:00:32.7447 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-]   File "/usr/lib/enigma2/python/Navigation.py", line 251, in dispatchEvent
04:00:32.7459 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-]   File "/usr/lib/enigma2/python/Components/ServiceEventTracker.py", line 80, in event
04:00:32.7465 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/WebkitHbbTV/plugin.py", line 252, in _cb_ait_detected
04:00:32.7480 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-]     self._app_info = self.get_autostart_application()
04:00:32.7485 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/WebkitHbbTV/plugin.py", line 307, in get_autostart_application
04:00:32.7491 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-]     reader.doDump()
04:00:32.7495 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/WebkitHbbTV/aitreader.py", line 95, in doDump
04:00:32.7504 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-]     print "Name  :", x["name"]
04:00:32.7508 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-]   File "/usr/lib/python2.7/site-packages/twisted/logger/_io.py", line 163, in write
04:00:32.7554 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-]   File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
04:00:32.7597 { E } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-05-01 04:00:32+0200 [-] UnicodeDecodeError: 'utf8' codec can't decode byte 0xb0 in position 21: invalid start byte
04:00:32.7598 [ E ] python/python.cpp:210 call [ePyObject] (PyObject_CallObject(<bound method Navigation.dispatchEvent of <Navigation.Navigation instance at 0x8fe6ab48>>,(21,)) failed)

atvcaptain, 2019-05-01:
hatte es schon vor paar wochen gefixt, hab das plugin noch mal gepuscht das es neu gebaut wird

jbleyel, 2019-05-01:
Wo genau hast du das gefixt?

Der Crash ist doch hier ->
https://github.com/oe-alliance/enigma2-plugin-extensions-hbbtv-webkit/blob/master/HbbTV/aitreader.py#L95

Zumindest wenn das der richtige Source ist.

atvcaptain, 2019-05-01:
https://github.com/oe-alliance/enigma2-plugin-extensions-hbbtv-webkit/commit/1afa89b71f69fb6685008c38b828bb13932dae28#diff-0ff2f320e6afa0a4d750e884f372bb17

das hier ist für mipsel gewesen

atvcaptain, 2019-05-01:
arm da gab es noch commits danach  das war der start

https://github.com/oe-alliance/oe-alliance-core/commit/eb844c6acb06217cde13c9ef3246e5dc125712e4#diff-2a2c9876d4f5949a6654e47883854f57

atvcaptain, 2019-05-01:
hatte auch damals eine mail an vu gesendet ob sie das rausnehmen können, aber wie immer null reaktion

jbleyel, 2019-05-01:
Schau mal mein crashlog.
Der Absturz ist nicht in der plugin.py sonder in der aitreader.py

01 04:00:32+0200 [-] File "/usr/lib/enigma2/python/Plugins/Extensions/WebkitHbbTV/aitreader.py", line 95, in doDump

atvcaptain, 2019-05-01:
aber der aufruf kommt von plugin.py sonst wird das nirgens verwendet

jbleyel, 2019-05-01:
Sorry ich BLIND ;-)

closed 2019-05-01 by jbleyel