oe-alliance/oe-alliance-core #496: Zgemma H9 Combo and Ci info

Marchella1983, 2019-05-01:
Please tell me where does CommonInterfaceAssignment draw this information, image or driver?
```
no module found
init modules
```

atvcaptain, 2019-05-01:
we have report this, its driver

Marchella1983, 2019-05-01:
I have already reported this, but the manufacturer writes the opposite
https://github.com/zgemma-star/meta-zgemma/issues/46

atvcaptain, 2019-05-01:
i speak when insert one modul  in the box and not working

Marchella1983, 2019-05-01:
now I haven’t inserted a module, and in CommonInterfaceAssignment the initialization is constantly going on
[image]
and why he shows two slots and not one

atvcaptain, 2019-05-01:
the comes from driver report two ci slots, but possible add e2 hack to hite one

Marchella1983, 2019-05-01:
there is one Ci slot on the H9 Combo and it will be correct if it outputs one slot and not two, please check this error

Marchella1983, 2019-05-01:
one more request, it can somehow be fixed
root@zgemmah9combo:~# cat /proc/bus/nim_sockets
NIM Socket 0:
        Type: DVB-T2
        Name: Si2168
        Has_Outputs: no
        Frontend_Device: 0
        I2C_Device: 0
        Supports_Blind_Scan: yes
NIM Socket 1:
        Type: DVB-S2X
        Name: Si2166D
        Has_Outputs: no
        Frontend_Device: 1
        I2C_Device: 0
        Supports_Blind_Scan: yes
root@zgemmah9combo:~# 
```
I wanted to see it like that
NIM Socket 0:
        Type: DVB-S2X
        Name: Si2166D
        Has_Outputs: no
        Frontend_Device: 1
        I2C_Device: 0
        Supports_Blind_Scan: yes
NIM Socket 1:
        Type: DVB-T2
        Name: Si2168
        Has_Outputs: no
        Frontend_Device: 0
        I2C_Device: 0
        Supports_Blind_Scan: yes
```

atvcaptain, 2019-05-01:
[image]

better :)

Marchella1983, 2019-05-01:
@atvcaptain fine, super ...

atvcaptain, 2019-05-01:
tuner sort order is not a issue, you like dvb-s first other user like dvb-c first  and so on, the output comes from the hardware and internel i2c bus slot, we dont add a e2 hack

closed 2019-05-01 by atvcaptain

Marchella1983, 2019-05-02:
hello, if you can then edit here
[image]
[image]

atvcaptain, 2019-05-02:
ok fix include

Marchella1983, 2019-05-16:
> we have report this, its driver

**Captain** Hello, please tell me what has been fixed in the new version of the H9 Combo driver?

I do not see that this problem is fixed with the constant initialization of the module even if it is not inserted in the Ci slot ...
[image]

atvcaptain, 2019-05-16:
no problem here now works with german moduls

Marchella1983, 2019-05-17:
If there is no module> no module found

and here always> init module

atvcaptain, 2019-05-17:
old driver working ?

Marchella1983, 2019-05-17:
not

Marchella1983, 2019-05-17:
inserted a module
[image]

removed the module
[image]

create restart gui and again module init
[image]

Marchella1983, 2019-05-17:
set Open Pli 7.1 rc and everything works correctly there
[image]
[image]

atvcaptain, 2019-05-17:
use pli

Marchella1983, 2019-05-17:
an excellent approach will get rid of the problem that exists and must be corrected...

for me, better atv I don't see a better image ...

atvcaptain, 2019-05-17:
[image]

ci+ 1.3 all fine

atvcaptain, 2019-05-17:
ac modul too

[image]

Marchella1983, 2019-05-17:
Why do you insert the module?

I'm talking about the fact that even without a module in CommonInterfaceAssignment constantly writes the init module instead of no module found

atvcaptain, 2019-05-17:
i understand no modul found

Marchella1983, 2019-05-17:
Captain look, this is a screen without a module on the ATV
[image]

and this with openpli
[image]

Marchella1983, 2019-05-17:
> for me looks all ok

Now do a restart of the gui and see what the information will be?

atvcaptain, 2019-05-17:
i see no problem here, i dont understand you what you mean

Marchella1983, 2019-05-17:
> i start box without modul
> 
> [image]
> 
> for me looks all ok

make a rectart Gui H9 Combo without a module and show the screen

atvcaptain, 2019-05-17:
for me ci works  https://youtu.be/5wy4wqeK0O0

Marchella1983, 2019-05-17:
yes you have no problems but you have ATV_6.3 I have 6.2
[image]

arn354, 2019-05-17:
then install 6.3 and check again - clean install, default skin.
there are no more changes to 6.2 enigma2 foreseen.

Marchella1983, 2019-05-17:
thanks, now everything has become clear, installed 6.3 and there are no more problems ...

once again, please apologize for your perseverance and your time spent on me ...

atvcaptain, 2019-05-17:
6.2 goes soon not for all to eol

Marchella1983, 2019-05-26:
Hello, Captain, again problems after yesterday's update
[image]

zgemma-star, 2019-05-31:
> 
> 
> ok fix include

The H9Twin has two CI and the reports from customers it seems Openatv only one shown one CI.

Marchella1983, 2019-05-31:
**@zgemma-star** Can I ask them to show a screenshot with two modules inserted in the Ci slot?