oe-alliance/oe-alliance-core #508: Some of the context lines in patches were ignored. This can lead to incorrectly applied patches.

Ormiwhite, 2019-07-02:
i try to build openatv for dreambox, i get this warning , if someone can please show me how to fix it.

WARNING: linux-dreambox-3.2-.7 do_patch: 
Some of the context lines in patches were ignored. This can lead to incorrectly applied patches.
The context lines in the patches can be updated with devtool:

    devtool modify <recipe>
    devtool finish --force-patch-refresh <recipe> <layer_path>

Then the updated patches and the source tree (in devtool's workspace)
should be reviewed to make sure the patches apply in the correct place
and don't introduce duplicate lines (which can, and does happen
when some of the context is ignored). Further information:
http://lists.openembedded.org/pipermail/openembedded-core/2018-March/148675.html
https://bugzilla.yoctoproject.org/show_bug.cgi?id=10450
Details:
Applying patch 0001-Revert-MIPS-Fix-potencial-corruption.patch
patching file arch/mips/mm/c-r4k.c
Hunk #1 succeeded at 642 (offset 6 lines).
Hunk #2 succeeded at 653 with fuzz 1 (offset 6 lines).
Hunk #3 succeeded at 666 with fuzz 1 (offset 7 lines).

Now at patch 0001-Revert-MIPS-Fix-potencial-corruption.patch
WARNING: linux-dreambox-3.2-.7 do_patch: 
Some of the context lines in patches were ignored. This can lead to incorrectly applied patches.
The context lines in the patches can be updated with devtool:

    devtool modify <recipe>
    devtool finish --force-patch-refresh <recipe> <layer_path>

Then the updated patches and the source tree (in devtool's workspace)
should be reviewed to make sure the patches apply in the correct place
and don't introduce duplicate lines (which can, and does happen
when some of the context is ignored). Further information:
http://lists.openembedded.org/pipermail/openembedded-core/2018-March/148675.html
https://bugzilla.yoctoproject.org/show_bug.cgi?id=10450

nickersk, 2019-07-02:
hi, you can safely ignore such warnings. if something goes really wrong, bitbake will throw error and stop the build. or if you want to fix such warning, you have to initialize OE build env and do what bitbake suggest you to do :-)

Ormiwhite, 2019-07-03:
thank's , can build only enigma2 and not all image ? i made some modification , i want to know if it's work , before build an image 

thnk's

closed 2019-07-03 by Ormiwhite

nickersk, 2019-07-03:
from topdir:
MACHINE=box; cd builds/openatv/release/box ; source env.source ; bitbake enigma2/enigma2.bb

Ormiwhite, 2019-07-03:
thnk's a lot brother it's work, but only one problem ;

↳ source env.source
check internet connection: Offline 

im connected to internet  and build run normally ...

it's okay with this ?

nickersk, 2019-07-03:
yes,it is ok.

daveHitchman, 2023-11-07:
One huge problem for me in using this devtool finish, I cant find anywhere anything that translates '<layer-path>' to something that actually means anything to me. What the blazes is the 'layer path' ???? Please, I am only English, I would expect to understand English, even Americans version of English, but this phrase might frankly be some martian language for all I can guess what the hell it means, there are no examples I can find. I have a patch that claims it needs 'fuzzing' which I 'assume' means that somewhere a line number might be a little out now for some reason - though heaven only knows what reason, it worked fine last week and I didnt change anything - but then the instructions seem not to produce any change that is useful, discernable or anything so I guess I need the finish command but what the hell is the layer path? I have been doing software since 1982 when it was a trs80 or acorn atom depemding on what was available and I can say 100% that things have gone backwards, more convoluted, less well documented and zero useful explanation

daveHitchman, 2023-11-07:
Oh ffks sake its even worse now ... no warning any more but the patch files havent changed!

Huevos, 2023-11-07:
> One huge problem for me in using this devtool finish, I cant find anywhere anything that translates '' to something that actually means anything to me. What the blazes is the 'layer path' ???? Please, I am only English, I would expect to understand English, even Americans version of English, but this phrase might frankly be some martian language for all I can guess what the hell it means, there are no examples I can find. I have a patch that claims it needs 'fuzzing' which I 'assume' means that somewhere a line number might be a little out now for some reason - though heaven only knows what reason, it worked fine last week and I didnt change anything - but then the instructions seem not to produce any change that is useful, discernable or anything so I guess I need the finish command but what the hell is the layer path? I have been doing software since 1982 when it was a trs80 or acorn atom depemding on what was available and I can say 100% that things have gone backwards, more convoluted, less well documented and zero useful explanation

Is this a question or a rant? Please try to ask a clear question stating the problems encountered along with background information like what you are trying to build, so we can help you with any problem you might be having.

daveHitchman, 2023-11-08:
In some respects I guess its both. The question is what does it mean when
it says <layer path>. I am applying a patch to a file that is somehow
created during the build of the recipe, even that is it seems some black
magic because if I ask the tools to tell me what is building it (it isnt
copied I can see) then it tells me it doesnt know, but if I clean the build
stuff away totally it does reappear so clearly something somewhere is. But
then the patch broke - no real idea why, it claimed it could fuzz it to
work but I wanted to clear that warning as I dont like a lot of warnings
hanging around as they mask the new ones. There is documentation that says
this devtool thing can do it but the last command is not clear how it is
actually used - or at least not clear to someone who has only been fighting
yocto recipes for a couple of weeks. However - the whole problem appears to
have cured itself again - the only intervention being to have cleared out
all the build stuff so something had got itself in a twist and has
untwisted itself from a clean build, that in itself worries me a little.

So the question is:
What does <layer path> mean in the contect of the 'devtool finish' command.
How do I find what 'layer' I am 'patching' in order to provide this path?

Thanks
Dave

atvcaptain, 2023-11-08:
layer path is whether bitbake recognizes your recipe when scanning, if you have stored your private recipes in meta-local/recipes-local/my folder/my.bb you don't have to worry about it the search mask will find your bb
it will be set here
https://github.com/oe-alliance/build-enviroment/blob/5.3/meta-local/conf/layer.conf#L5-L6

daveHitchman, 2023-11-08:
Fantastic thanks, yes I have my own custom in a directory just under
sources along with the meta-imx, meta-python, poky directories. Seems to
work. The warning has returned this morning - oh well :)