oe-alliance/oe-alliance-core #515: update oe-alliance-core ok , no for enigma2 ?

Ormiwhite, 2019-08-21:
build openspa openbh work well,  now i try to update image, so i made  make update , i get update only for oe-alliance-core , i don't get update for enigma2  ? 
to get all update i should remove build folder !

nickersk, 2019-08-21:
that is ok, e2 is built from enigma2.bb which is fetching latest git.

closed 2019-08-21 by nickersk

Ormiwhite, 2019-08-21:
ok for oea-core , but for e2 i know there some change ,but i don't get theme, i don't know why,  i should remove build folder to get last update.

nickersk, 2019-08-21:
image gets updates index from feed uri, you have to install e2 ipk manually ... or setup your own ipk feeds and change uris in opkg confs

Ormiwhite, 2019-08-21:
sorry my friend , i think im not clear about this issue , 
ok, i build my image from branch 4.3 ok .
the build is ok , i setup my local feed ok.
enigma2 it's uptodate  ;

now if you made some upgrade or change with enigma2.....

Openhdf enigma2 it's here :  https://github.com/openhdf/enigma2
if developer make some change update with e2 , and i want to get this  last change ,   so i do "make update" , i build new image all ok, but enigma2  don't change ,  stay as  first time  i made build , i don't get last change for enigma2 ?

nickersk, 2019-08-21:
you must rebuild image with MACHINE=vusolose make image, then copy enigma2 ipk from deploy/ipk dir to box and after install it on box with opkg install enigma2... i cant write it more clearer ;-)

Ormiwhite, 2019-08-21:
my friend , it's me im not clear, not about you sorry, i know how to make update , it's not my problem  i build image and feed at local, the issue is when i build image and i want to make update ( after u update enigma2) , the enigma2 stay at old version, the issue it's at build i don't get last enigma2 change, not how to update my box, it's clear and easy .. and i do it with telnet i have working local feed  " init 4 opkg update opkg upgrade sync init 3"  ... thnks

arn354, 2019-08-21:
there is no "make update" needed to build a new commit in the enigma2 repository...
in case there is a change on the defined url of enigma2 repository in distro.conf you build from - this is automatically fetched!
Check which url you build from.
Check the build log for enigma2-6.3 commit-hash.
Check your build directory ...deploy/ipk for enigma2 package - is it there.
Check your feed server for enigma2 package - is it there.
Check your box - which feed url is used, which package version of enigma2 sees opkg from your box.

Ormiwhite, 2019-08-21:
thanks 
please how to do this : Re-Run packages.gz

nickersk, 2019-08-22:
bitbake package-index

Ormiwhite, 2019-08-22:
hi , now i will show you my problem, i build new image today ok, 
there is commit new commit here:

https://github.com/OpenSPA/dvbapp2/commit/b43f3849b485ab8b8aa5af8fee333890cfbdea40#diff-7b38577a2823523995b7c4770f7c73c4R16

now in my box i don't get this update:

```
↳ uncompyle6 PowerLost.pyo 
# uncompyle6 version 3.3.4
# Python bytecode 2.7 (62211)
# Decompiled from: Python 2.7.16 (default, Apr  6 2019, 01:42:57) 
# [GCC 8.3.0]
# Embedded file name: /usr/lib/enigma2/python/Screens/PowerLost.py
# Compiled at: 2019-08-17 07:04:12
from Screen import Screen
from MessageBox import MessageBox
from Components.config import config
import Screens.Standby
from boxbranding import getMachineBrand, getMachineName

class PowerLost(Screen):

    def __init__(self, session):
        Screen.__init__(self, session)
        self.showMessageBox()

    def showMessageBox(self):
        if config.usage.boot_action.value == 'normal':
            message = _('Your %s %s was not shutdown properly.\n\nDo you want to put it in %s?') % (
             getMachineBrand(), getMachineName(), config.usage.shutdownNOK_action.value)
            self.session.openWithCallback(self.MsgBoxClosed, MessageBox, message, MessageBox.TYPE_YESNO, timeout=60, default=True)
        else:
            self.MsgBoxClosed(True)

    def MsgBoxClosed(self, ret):
        if ret:
            if config.usage.shutdownNOK_action.value == 'deepstandby' and not config.usage.shutdownOK.value:
                self.session.open(Screens.Standby.TryQuitMainloop, 1)
            elif not Screens.Standby.inStandby:
                self.session.open(Screens.Standby.Standby)
        self.close()
# okay decompiling PowerLost.pyo

```

it's stay at the last time i remove build folder  here  2019-08-17 ? after that i don't get the last commit ?!

arn354, 2019-08-22:
again... check your build-log or build directory .../deploy/ipk - provide the filename of enigma2_6.3 ipk e.g.:
teambluebuilder@teamblue:~/teamblue-6.3-DEV-arm/build-enviroment/builds/teamblue/DEV/gb7252/tmp/deploy/ipk/gbquad4k$ ls |grep enigma2_6.3
enigma2_6.3+git17690+236ba9c-r6_gbquad4k.ipk

Which package is actually build on your buildserver?

Huevos, 2019-08-22:
If for some reason a package is not getting rebuild (maybe due to a forced push) just clean the recipe and try again.

e.g. "cd" to machine folder and then run the following...

openvix@ubuntu01:~/5.2/builds/openvix/experimental/osmio4k$ . env.source
openvix@ubuntu01:~/5.2/builds/openvix/experimental/osmio4k$bitbake -c clean <package name>

Then build again.

Ormiwhite, 2019-08-23:
hi 
@arn354 i have this:

enigma2-7.4+gitAUTOINC+b0e52f018a-r0

@Huevos
so i do this:
bitbake -c cleanall enigma2
0: enigma2-7.4+gitAUTOINC+b0e52f018a-r0 do_cleansstate - 0s (pid 10652)

i build again , and update my box with telnet:

 opkg install --force-reinstall enigma2

Reinstalling enigma2 (7.4+git323+b0e52f0-r0) on root.
Installing enigma2 (7.4+git323+b0e52f0) on root.
Configuring enigma2.

so it's not me,  enigma2 at same version:

```

# Embedded file name: /usr/lib/enigma2/python/Screens/PowerLost.py
# Compiled at: 2019-08-23 09:14:11
from Screen import Screen
from MessageBox import MessageBox
from Components.config import config
import Screens.Standby
from boxbranding import getMachineBrand, getMachineName

class PowerLost(Screen):

    def __init__(self, session):
        Screen.__init__(self, session)
        self.showMessageBox()

    def showMessageBox(self):
        if config.usage.boot_action.value == 'normal':
            message = _('Your %s %s was not shutdown properly.\n\nDo you want to put it in %s?') % (
             getMachineBrand(), getMachineName(), config.usage.shutdownNOK_action.value)
            self.session.openWithCallback(self.MsgBoxClosed, MessageBox, message, MessageBox.TYPE_YESNO, timeout=60, default=True)
        else:
            self.MsgBoxClosed(True)

    def MsgBoxClosed(self, ret):
        if ret:
            if config.usage.shutdownNOK_action.value == 'deepstandby' and not config.usage.shutdownOK.value:
                self.session.open(Screens.Standby.TryQuitMainloop, 1)
            elif not Screens.Standby.inStandby:
                self.session.open(Screens.Standby.Standby)
        self.close()
# okay decompiling PowerLost.pyo

```

nickersk, 2019-08-23:
check which branch is e2 recipe fetching

arn354, 2019-08-23:
he should ask openspa.
His problems started once openspa switched to 7.4 .
https://github.com/OpenSPA/dvbapp2/commit/a3968918392cd88298795e0b81de0d17e28de3e1