oe-alliance/oe-alliance-core #517: [security] Insecure system defaults: root has empy password

drwetter, 2019-08-21:
An attacker can own (=take over) the machine from the local network. ssh root@<IPADDRESS> or telnet` as root`` gives one complete access. 

This is IMO a huge security problem either in the LAN or if somewhere else exposed. 

Also at least telnet and vsftp are affected.

See also  https://github.com/openatv/enigma2/issues/1361

opfans, 2019-09-15:
you can set a root password youself.

drwetter, 2019-09-15:
Please read again the title and the consult https://en.wikipedia.org/wiki/Secure_by_default.

drwetter, 2019-09-19:
You guys may also want to read https://en.wikipedia.org/wiki/Mirai_(malware)

Schimmelreiter, 2019-09-19:
https://github.com/oe-alliance/oe-alliance-core/commit/2408e713a25a0b4b43890af5d29a47939c4bfb7d

atvcaptain, 2019-10-06:
yes we know about this, but less helping hands, so all need very long
in the next image user must set a password to have access remote to the box its on the todo list

TwolDE2, 2020-01-26:
Is there any way of reverting to a null (blank) password as in previous versions?

AbuBaniaz, 2020-03-06:
4.4 branch requires a password for FTP. This can be set by telnet/SSH or in the Network/Password menu.

 Is it OK to close this issue now?

closed 2020-03-06 by arn354