oe-alliance/oe-alliance-core #528: /proc/net/ip_tables_names empty

drwetter, 2019-08-26:
Hi,

how come ip(6)tables complains about a missing filter and other modules:

```
root@sf8008:/lib/modules/4.4.35# iptables -L -nv
iptables v1.6.2: can't initialize iptables table `filter': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
root@sf8008:/lib/modules/4.4.35# iptables -t nat -L -nv
iptables v1.6.2: can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
root@sf8008:/lib/modules/4.4.35# iptables -t mangle -L -nv
iptables v1.6.2: can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
```

```
root@sf8008:/lib/modules/4.4.35# cat /proc/net/ip_tables_names 
root@sf8008:/lib/modules/4.4.35# 
```
It looks to me [1] like at least ``filter`` should be there (https://github.com/oe-alliance/oe-alliance-core/blob/03e47b47e1503cf22fae97aa3889024d8fba2018/meta-brands/meta-octagon/recipes-linux/linux-octagon-4.4.35/sf8008/defconfig, see ``CONFIG_NETFILTER=y`` and ``CONFIG_IP_NF_IPTABLES=y``)

I also couldn't find any external kernel modules, nor installed:

```
root@sf8008:/lib/modules/4.4.35# opkg list | grep kmod | grep ipt
root@sf8008:/lib/modules/4.4.35# 
```

System: see above. Unpacked, installed, upgraded to 6.3.

[1] No clue how kernels are being build here and got onto my system.

arn354, 2019-08-27:
how a box gets build: https://github.com/openatv/enigma2/blob/6.3/README.md
yes - seems defconfig is not complete for that specific box. The manufacturer should load that defconfig and recreate it for usage with iptables .
You probably saw some "unsatisfied recommendation" s on installation - which no longer seem to be recommendations but dependencies for iptables 1.6.2 -no idea, if upstream bitbake receipe is not correct. iptables comes from openembedded - https://github.com/openembedded/openembedded-core/commits/master/meta/recipes-extended/iptables - it got updated in the meantime.
We use https://github.com/oe-alliance/build-enviroment/commit/91e1a52a0d94624827064ac3a73f58750c194d12 for openembedded-core as base for oe-alliance 4.3 (openATV 6.3). A later version of openembedded-core is used for branch nextp3 of oe-a - https://github.com/oe-alliance/build-enviroment/tree/nextp3 - that one should contain the later version of iptables and will be in future oe-a 4.4 builds (e.g. openatv 6.4 or 7).

teamblue-e2, 2019-08-31:
after updating defconfig related to iptables for gb7252:
```
root@gbquad4k:~# iptables -L -nv
Chain INPUT (policy ACCEPT 31 packets, 3684 bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain OUTPUT (policy ACCEPT 19 packets, 10466 bytes)
 pkts bytes target     prot opt in     out     source               destination

root@gbquad4k:~# iptables -A INPUT -i lo -j ACCEPT
root@gbquad4k:~# iptables -A OUTPUT -o lo -j ACCEPT
root@gbquad4k:~# iptables -A INPUT -p tcp --dport 22 -j ACCEPT
root@gbquad4k:~# iptables -A INPUT -p tcp --dport 80 -j ACCEPT
root@gbquad4k:~# iptables -A INPUT -p tcp --dport 443 -j ACCEPT
root@gbquad4k:~# iptables -A INPUT -j DROP

root@gbquad4k:~# iptables -L -nv
Chain INPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination
    0     0 ACCEPT     all  --  lo     *       0.0.0.0/0            0.0.0.0/0
  181 13160 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:22
    0     0 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:80
    0     0 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:443
    1    80 DROP       all  --  *      *       0.0.0.0/0            0.0.0.0/0

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain OUTPUT (policy ACCEPT 7 packets, 1008 bytes)
 pkts bytes target     prot opt in     out     source               destination
    0     0 ACCEPT     all  --  *      lo      0.0.0.0/0            0.0.0.0/0
root@gbquad4k:~#

```

teamblue-e2, 2019-08-31:
```
root@gbquad4k:~# cat /proc/net/ip_tables_names
mangle
nat
filter
```

the proc is only filled after it is used - e.g.:
iptables -L -nv
iptables -t nat -L -nv
iptables -t mangle -L -nv

drwetter, 2019-08-31:
@teamblue-e2 : You lost me or maybe I should have added that there's no kernel module for iptables:

```
root@sf8008:~# lsmod | grep -E "^ip|^^xt|^nf"
root@sf8008:~# 
```

I also tried to add rules 9see error msg).

```
root@sf8008:~# iptables -A INPUT -i lo -j ACCEPT
iptables v1.6.2: can't initialize iptables table `filter': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
root@sf8008:~# 
```

So, it's the iptables version doesn't matter. For the LEDE/OpenWRT systems some modules are in the repository:

```
root@LEDE:~# opkg list | grep 'kmod-ip' | wc -l
61
root@LEDE:~# 
```

Is there any way just installing a kernel or get the right modules without installing a cross build environment?

teamblue-e2, 2019-08-31:
seems you did not understand... once there is a proper defconfig for your box - then iptables will work. updated defconfig for gb7252 will be commited later.

drwetter, 2019-08-31:
I am familiar with Linux, but I have zero idea how deployment works for my box and your terminology (what is ``gb7252 `` -- a build?).

If not then how a kernel configuration in place (=``defconfig`` IMO) will make "iptables work" for me?

As said despite the info @arn354 pointed me to, I am not eager to build the kernel myself.

teamblue-e2, 2019-08-31:
sf8008 is your box - someone who is able to build and check your box should modify the defconfig and provide a pull request once change does not brick the box:
https://github.com/oe-alliance/oe-alliance-core/blob/4.3/meta-brands/meta-octagon/recipes-linux/linux-octagon-4.4.35/sf8008/defconfig

gb7252 is a machinebuild for gbquad4k and gbue4k - that one will be updated soon:
https://github.com/oe-alliance/oe-alliance-core/blob/4.3/meta-brands/meta-gigablue/recipes-linux/linux-gigablue-4.1.20/gb7252/defconfig

teamblue-e2, 2019-08-31:
a regular image build for openatv 6.3 using oe-a 4.3 is documented there: https://github.com/openatv/enigma2/blob/6.3/README.md

drwetter, 2019-08-31:
I read the latter, thanks. But as said I not eager to build myself the kernel. Correction: Building is not the problem. I have no clue about the boot procedure and don't know whether I can brick the box. And the fact that the kernel configuration here (defconfig) said netfilter is enabled (CONFIG_IP_NF_IPTABLES=y etc.) but on my box it is not, makes me not confident the whole thing will work without any problems, means it's costs too much time.

teamblue-e2, 2019-08-31:
again - the defconfig of sf8008 is not complete - it lacks further required parts for iptables support. Otherwise you will not get your missed modules/functions. "Someone" has to change it for Octagon SF8008 - GigaBlue will change it for GigaBlue devices (gb7252, gbmv200, gb7356,...).

arn354, 2019-08-31:
for GigaBlue see:
gb7252 - https://github.com/oe-alliance/oe-alliance-core/commit/4bb0dfb8a270ba4be3ad85f9b52bfc9891f6a14a
gbmv200 - https://github.com/oe-alliance/oe-alliance-core/commit/0b7b07463a60a831c8c7e49c3c306ec0f1a99fd6
...

arn354, 2019-09-01:
GigaBlue gb7252 and gbmv200 is now working ipv4/ipv6 - see below

```
root@gbquad4k:~# ip6tables -L -nv
Chain INPUT (policy ACCEPT 14 packets, 3717 bytes)
 pkts bytes target     prot opt in     out     source               destination
    0     0 ACCEPT     tcp      *      *       ::/0                 ::/0                 state NEW tcp dpt:22
    0     0 ACCEPT     all      *      *       ::/0                 ::/0                 state RELATED,ESTABLISHED

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain OUTPUT (policy ACCEPT 1 packets, 117 bytes)
 pkts bytes target     prot opt in     out     source               destination
root@gbquad4k:~#

```

```
root@gbquad4k:~# lsmod | grep -E "^ip|^^xt|^nf"
nf_conntrack_ipv6       7970  2
nf_defrag_ipv6         16542  1 nf_conntrack_ipv6
ip6table_filter          859  1
ip6_tables             10196  1 ip6table_filter
nf_conntrack_ipv4       7603  2
nf_defrag_ipv4          1175  1 nf_conntrack_ipv4
xt_state                 909  4
nf_conntrack           60046  3 xt_state,nf_conntrack_ipv4,nf_conntrack_ipv6
xt_tcpudp               2629  4
iptable_filter           912  1
ip_tables              10917  1 iptable_filter
root@gbquad4k:~#

```

closed 2019-10-05 by atvcaptain

drwetter, 2019-10-06:
Did I miss something or why has this been closed now? If this a giga* only repo could you pls tell me where do I get a binary build for sf8008 from with proper iptables support?

atvcaptain, 2019-10-06:
For me, it looked like everything was ok, long time no more infos here
i  have not see arn not merge the change to all to all others oem´s

next days changes now include fpr sf8008 too

https://github.com/oe-alliance/oe-alliance-core/commit/6e4d7fb68839be0b7c8c8938464b6110acd78719

teamblue-e2, 2019-10-06:
everything provided on a silver spoon - no one wanted to support sf8008.
not the manufacturer (which should have been asked)
not the original requester (who had all informations)
so it was fine to close ;)

drwetter, 2019-10-06:
What kind of statement is that? You guys seem to work in a beginner level here. 

Now I start to understand why the products are so broken from the security perspective.

arn354, 2019-10-06:
...please help or move further.
why should Brand A fix stuff for Brand B?
Just whining i want, i want and doing nothing is the far better approach?

betacentauri, 2019-10-07:
A firewall on the box doesn't really make it more secure. It still uses in most cases ancient kernels (which we cannot change).

Huevos, 2019-10-07:
Why would anyone want a firewall on the box? It shouldn't be exposed to the public internet.