oe-alliance/oe-alliance-core #530: Crash selecting any language other than German or English on receivers with small flash

ViX-Sicilian, 2019-08-28:
Why have Languages been removed for small flash receivers? Leaving only English US, UK and German?

This causes a crash with OpenViX due to the languages now missing. How can this be corrected?

OpenATV have removed languages, but there are many users that use other languages.

TwolDE2, 2019-08-28:
More difficult, but it would be better to pick up the "small flash" situation in Language.py (BoxBranding??) and as an option, delete languages after selection

arn354, 2019-08-28:
Q: Why have Languages been removed for small flash receivers? 
A: 
1 Because the image is no longer flashable, if the image size is not reduced. (CFE issue)
2 Because the image is no longer usable, if rootfs is full 99%. (Box Flashsize issue)
3 Over the years with every update of openembedded etc. imagesize grows and grows. A lot of additional tools, drivers etc. already got removed from default installation. So there are not much choices anymore to reduce image size.

So you could either prevent the crash in Enigma2 adapting Language-Selection or you could exclude openvix for the change in meta-local - but then you have to fix the image size for the boxes in question - one by one or no longer build them.

arn354, 2019-08-28:
In short - time will come to drop support for smallflash boxes and boxes with CFE issue in future OE-A versions.

RobvanderDoes, 2019-08-28:
But surely a 0.5GB flash box like the ET8000 isn't a 'small flash' box.

arn354, 2019-08-28:
but thats a box with CFE limit - try to flash a image with rootfs size of 120MB for example..... so hen and egg...

RobvanderDoes, 2019-08-28:
I don't have an image that big.
But I do have a crash thanks to missing system files.

original-birdman, 2019-08-28:
> So you could either prevent the crash in Enigma2 adapting Language-Selection or you could exclude openvix for the change in meta-local - but then you have to fix the image size for the boxes in question - one by one or no longer build them.

Or you could make the additional languages installable packages.

atvcaptain, 2019-08-28:
add for vix only all, or fix the e2 crash atv dont crash

arn354, 2019-08-28:
then correct it - the way you prefer it, for every box you build.

TwolDE2, 2019-08-28:
Guys this is a user issue - people have paid money for the hardware and expect to be able to install their local language - the change is a good idea thats trying to fix a real issue, but you cannot deliver an image where a user cannot install their language - if you really don‘t agree then delete German language from the list and see how you handle that situation.

arn354, 2019-08-28:
just look into enigma2 of openatv - yes.
e.g. one part : https://github.com/openatv/enigma2/commit/0978f6475050a02714d05d00f4bf8b611ceee292#diff-95ad66c1e9f7cd7c5f4da100d617c264

atvcaptain, 2019-08-28:
??  the change is old   https://github.com/oe-alliance/oe-alliance-core/commit/f7e7f8dcb878b0e9e0fbfc12ee7d13edeaaa6e8f#diff-eb5dd480e1d22d5068c09c2b273803f2

i have change it for  12 months , user can load missing language from feed not big problem,

TwolDE2, 2019-08-28:
Great! Where in feeds do you download this?

TwolDE2, 2019-08-28:
OK - thanks - appreciate the feedback

TwolDE2, 2019-08-28:
....... and obviously I can do this in the initial setup?

closed 2019-10-03 by atvcaptain