oe-alliance/oe-alliance-core #532: [Alien 5] Problems

freelancers19, 2019-09-23:
- OpenATV 6.3 does not start. It blocks on "Amiko" 
- Sometimes on 4k channels, the decoder with openatv 6.2, crashes.
- in deepstanby the LED is blue and does not turn red.

nickersk, 2019-09-23:
any boot logs?

freelancers19, 2019-09-23:
With OpenATV 6.3; No logs. The "/home/root" folder is empty.
How can I enable the log at startup?

nickersk, 2019-09-23:
if there is nothing usable in /var/log , then we will need to see boot log from console ...

freelancers19, 2019-09-23:
how can I see the log from the console?

nickersk, 2019-09-23:
seems like console port is somewhere on mainboard ...

nickersk, 2019-09-24:
as i can see no more possibilities to fix this for now i will close this  thread. you can try to ask manufacturer for support (so we can integrate needed changes to image). thank you

closed 2019-09-24 by nickersk