oe-alliance/oe-alliance-core #536: FFMPEG zip Error !

thawtes, 2019-09-28:
hi,
ffmpeg error message 
[http @ 0x4463b0] Compressed (gzip) content, need zlib with gzip support
http://sampledomain.com/playlist.m3u8: Function not implemented

pls fix @atvcaptain @nickersk

nickersk, 2019-09-28:
looks like zlib is compiled with gz support... can you please paste log part, where this error occured?

thawtes, 2019-09-29:
> 
> 
> looks like zlib is compiled with gz support... ok, this comes from http.c will try to find why ...
thanks...

dear @nickersk 
I found the problem !

problem caused by **libavformat58** package
I installed the previous version of this package and the problem is solved!

so what are we supposed to do now?
I leave it to you to fix it from now on...

nickersk, 2019-09-29:
there is check for compile flags 16,17 of zlib, from manual: 16: NO_GZCOMPRESS — gz* functions cannot compress (to avoid linking deflate code when not needed)17: NO_GZIP — deflate can't write gzipstreams, and inflate can't detect and decode gzip streams (to avoid linking crc code)
this change wil affect whole built system... maybe we can just disable zlib like it was in previous versions of ffmpeg and is in OE recipe...

nickersk, 2019-09-29:
what are the minor version numbers of libavformat that works? current is libavformat 58. 29.100 you can find is simply by running ffmpeg command on cmdline without any parameters...

thawtes, 2019-10-01:
thanks, everything is working normally

closed 2019-10-01 by thawtes