oe-alliance/oe-alliance-core #537: nfs-utils 2.4.1

atvcaptain, 2019-10-02:
have you fix this issue before i start to search

BB_VERSION           = "1.43.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "osmio4kplus"
DISTRO               = "openatv"
DISTRO_VERSION       = "6.4"
TUNE_FEATURES        = "aarch64"

| gcc  -DHAVE_CONFIG_H -I. -I../../support/include  -isystem/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/nfs-utils/2.4.1-r0/recipe-sysroot-native/usr/include -D_GNU_SOURCE -D_GNU_SOURCE -isystem/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/nfs-utils/2.4.1-r0/recipe-sysroot-native/usr/include -march=native -O2 -pipe -Os -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/nfs-utils/2.4.1-r0=/usr/src/debug/nfs-utils/2.4.1-r0                      -fdebug-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/nfs-utils/2.4.1-r0=/usr/src/debug/nfs-utils/2.4.1-r0                      -fdebug-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/nfs-utils/2.4.1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/nfs-utils/2.4.1-r0/recipe-sysroot-native=  -Wno-error=format-overflow -c -o testlk-testlk.o `test -f 'testlk.c' || echo './'`testlk.c
| gcc: error: unrecognized command line option ‘-fmacro-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/nfs-utils/2.4.1-r0=/usr/src/debug/nfs-utils/2.4.1-r0’
| Makefile:455: recipe for target 'testlk-testlk.o' failed
| make[2]: *** [testlk-testlk.o] Error 1

nickersk, 2019-10-02:
i havent got this error, will try to build ...

atvcaptain, 2019-10-02:
get next issue
 cc1: error: unknown value 'native' for '-march'
| cc1: note: valid arguments are: armv8-a armv8.1-a armv8.2-a armv8.3-a armv8.4-a armv8.5-a
| Assembler messages:
| Error: unknown architecture `native'
| 
| Error: unrecognized option -march=native

i have remove temp the -march option in the site.conf

nickersk, 2019-10-02:
this is strange ... i have no errors like this

nickersk, 2019-10-02:
1st issue i think appears because of gcc7 second maybe too ..

atvcaptain, 2019-10-02:
same machine ?
so fast and all ready wow

nickersk, 2019-10-02:
its laptop in fact. dell g3 ..

atvcaptain, 2019-10-02:
about -march  https://gcc.gnu.org/onlinedocs/gcc/ARM-Options.html

nickersk, 2019-10-02:
my site.conf: SCONF_VERSION = "1"
BB_NUMBER_THREADS = "14"
PARALLEL_MAKE = "-j 12"
BUILD_OPTIMIZATION = "-march=native -O2 -pipe"
DL_DIR = "/home/sources"
INHERIT += "rm_work"

nickersk, 2019-10-02:
and it works without any problems ...

atvcaptain, 2019-10-02:
good for you :)  in 1 h i have more infos

nickersk, 2019-10-02:
stupid question, but have you installed all prereqs installed as stated here: https://www.yoctoproject.org/docs/2.7.1/ref-manual/ref-manual.html#required-packages-for-the-build-host

atvcaptain, 2019-10-02:
now ok
1) i remove  SRC_URI_remove = "file://0001-Don-t-build-tools-with-CC_FOR_BUILD.patch"
2) remove -march=native

nickersk, 2019-10-02:
ahh now i remember... exactly this caused error or newer oses. doing rebuild...

closed 2019-10-02 by atvcaptain

nickersk, 2019-10-03:
that patch however is completly wrong imho. i test another approach, will commit later if success.

atvcaptain, 2019-10-03:
at the moment next build issue i seach i open a new case

nickersk, 2019-10-03:
https://github.com/oe-alliance/oe-alliance-core/commit/6b9e0ddf3f8bc9e89eb875f5a780967197ef8c3c